TheSpaghettiDetective/OctoPrint-Obico #5: Error while installing

EpicLPer, 2019-03-18:
I'm getting the following error while trying to install the plugin:
`Installiere Plugin "The Spaghetti Detective (Beta)" von https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip --no-cache-dir
Collecting https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip
Downloading https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip (2.8MB)
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-Iy00sA-build/
Complete output from command python setup.py egg_info:
Found packages: ['octoprint_thespaghettidetective_beta']
running egg_info
error: Invalid distribution name or version syntax: The-Spaghetti-Detective-Beta--0.4.0
----------------------------------------
You are using pip version 9.0.3, however version 19.0.3 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output`

Log Output:
`2019-03-18 19:36:23,440   /home/pi/oprint/bin/python2 -m pip install https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip --no-cache-dir
2019-03-18 19:36:26,474 > Collecting https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip
2019-03-18 19:36:27,591 > Downloading https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip (2.8MB)
2019-03-18 19:36:33,662 > Complete output from command python setup.py egg_info:
2019-03-18 19:36:33,663 > Found packages: ['octoprint_thespaghettidetective_beta']
2019-03-18 19:36:33,663 > running egg_info
2019-03-18 19:36:33,663 > error: Invalid distribution name or version syntax: The-Spaghetti-Detective-Beta--0.4.0
2019-03-18 19:36:33,663 >
2019-03-18 19:36:33,664 > ----------------------------------------
2019-03-18 19:36:33,666 ! Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-hdttbU-build/
2019-03-18 19:36:33,666 ! You are using pip version 9.0.3, however version 19.0.3 is available.
2019-03-18 19:36:33,666 ! You should consider upgrading via the 'pip install --upgrade pip' command.
2019-03-18 19:44:59,618   /home/pi/oprint/bin/python2 -m pip install https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip --no-cache-dir
2019-03-18 19:45:03,673 > Collecting https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip
2019-03-18 19:45:04,791 > Downloading https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/beta.zip (2.8MB)
2019-03-18 19:45:17,520 ! Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-Iy00sA-build/
2019-03-18 19:45:18,025 > Complete output from command python setup.py egg_info:
2019-03-18 19:45:18,033 > Found packages: ['octoprint_thespaghettidetective_beta']
2019-03-18 19:45:18,034 > running egg_info
2019-03-18 19:45:18,035 > error: Invalid distribution name or version syntax: The-Spaghetti-Detective-Beta--0.4.0
2019-03-18 19:45:18,036 >
2019-03-18 19:45:18,037 > ----------------------------------------
2019-03-18 19:45:18,540 ! You are using pip version 9.0.3, however version 19.0.3 is available.
2019-03-18 19:45:18,541 ! You should consider upgrading via the 'pip install --upgrade pip' command.`

kennethjiang, 2019-03-18:
Thank you @EpicLPer for reporting this issue. It is fixed with #4 . Please try to re-install it and it should work.

closed 2019-03-18 by kennethjiang

EpicLPer, 2019-03-18:
> Thank you @EpicLPer for reporting this issue. It is fixed with #4 . Please try to re-install it and it should work.

Thanks, yeah now it works :)