TheSpaghettiDetective/OctoPrint-Obico #6: Webcam feed connection picks up new token but not websocket connection

kennethjiang, 2019-03-19:
Steps to reproduce:

1. Set up the plugin with one token.
2. Change it to another token.
3. BUG: webcam feed will start to use the new token to connect to server. But websocket connection is stuck with the old token.

Fix: disconnect and reconnect websocket when a new token is set.

psfales, 2019-03-19:
I don't know whether it's related or not, but when setting up the second account, it hung with the "spinning dots" in step 4 (waiting for camera connection).   I waited at least 5 minutes with no change, but when I went to the printers page, the printer was there (but without the image metadata that it, presumably, never got in step 4)

kennethjiang, 2019-03-23:
@psfales Sorry for missing your comments. The problem you described could be related to this issue. Are you still experiencing this problem after (presumably) restarting OctoPrint?

psfales, 2019-03-23:
@kennethjiang, thanks - my issue was resolving by restarting OctoPrint.   My March 23 comment was just intended to (possibly) help someone else reproduce the problem

kennethjiang, 2019-03-23:
Good to know! Thanks @psfales !

kennethjiang, 2021-01-25:
closing it as 1.5.0 is using a new way by removing the use of token.

closed 2021-01-25 by kennethjiang