TheSpaghettiDetective/OctoPrint-Obico #10: webcam.snapshot vs webcam.stream url [FR]

arhi, 2019-03-23:
Looking at the plugin source you are using stream value from webcam octoprint settings to fetch data. The way I understand this works you extract images from that stream right, maybe it would be possible to use snapshot url instead stream url?

why?

For example, on all my printers I tend to not use local camera and streamer but I use IP camera directly connected to the lan. So my webcam stream for e.g. looks like this:

rtsp://user:pass@192.168.1.11/iphone/11

and my snapshot url looks like

http://user:pass@192.168.1.11/tmpfs/snap.jpg

and cause of this:

- RTSP is not supported by most of the browsers so I can't view the stream in browser (I do using external app like VLC or PotPlayer) .. and most IP cameras support RTSP, only few rare ones support webm and similar streaming protocols

- snapshot is much higher resolution image then any video

From the quick test I'm making, TSD is not supporting RTSP neither :( and is ignoring snapshot url. Would be cool features to
 - support RTSP
 - support snapshot url 

furthermore, there's interesting plugin "Octolapse" that will create "better" timelapse that can be configured to for e.g. move the head and bed so that you get "perfect position to take image" .. interaction with it and fetching image taken by octolapse would probbly make detection easier and allow for use of snapshot url - so IP cameras are then fully supported :)

kennethjiang, 2019-03-23:
Moved this issue to this repo 'coz I believe it's more about the plugin than the server.

Actually TSD plugin does honor snapshot. It checks for snapshot url setting before moving on to stream url: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/ec5ba26957a84ae32c6233fe5908a4520d19e009/octoprint_thespaghettidetective_beta/webcam_capture.py#L19

But the current code to grab snapshot or cam stream (it's actually mjpeg, rather than a real streaming protocol) is very rudimentary. We can definitely use some help on enhancing it to work with users who don't use `mjpeg-streamer` recommended by OctoPrint.

And we'd LOVE to have a way to support RTSP or similar cam streaming protocol. Currently one big usability issue is the frame rate for our cam streaming. It's at pathetic 1 frame per 10 seconds. The reason why we can't make it much higher is that we are basically sending jpeg pics to the server, rather than a real video stream, which cost *a lot of* bandwidth. If the plugin sends it faster than 1 fps, it would mean a constant 1+Mbps bandwidth consumption which would cost a lot of connection issues for users within limited internet bandwidth.

Will love to brainstorm on a solution to add video encoding capability to this plugin so that it can stream video, not just jpeg pics. We briefly tested `ffmpeg` and found it would grind the Pi to a halt even just for a modest fps. We will love to get help from people who are more experienced than us on video encoding/streaming.

arhi, 2019-03-23:
Well, I'm running a server in LAN now :D :D :D :D so I don't have bw problem :D .. but yes, I agree, shooting bunch of streams to your server is not feasible for now ... maybe splitting some workload between client and server would work, dunno... 

I was not looking at the code much (I kinda dislike python so try to avoid it when I can :D ) so I seen only the streaming url and not the snapshot one :D ... I assume plugin sends this url to server and that server pulls this data? If I understand your message, the plugin is creating a snapshot (from stream or from snapshot url) and sending image by image to server for analysis?

kennethjiang, 2019-03-23:
Counter-intuitively, the bottleneck for bandwidth is not with the server, it's with the user's home network. This is the lesson I learned from my other [OctoPrint plugin](https://plugins.octoprint.org/plugins/anywhere/).

Yes the plugin pulls snapshots to memory and then sends them to server. The server usually can't directly access the urls because most of them are private address and behind firewalls.

arhi, 2019-03-24:
would be great if plugin would have a "test" button like there's one on the webcam & timelapse settings tab near url's so we can see that plugin properly pulled the config info (does plugin know to update info when config is changed in webcam settings?) .. attm there's just a test button to test if api token is valid. for e.g. I currently have no image on the server and I have no clue why

arhi, 2019-03-24:
this actually works exactly as  I wanted :D using snapshot url, it was issue #12  that was making problem for me... support for RTP would be cool but taking into account the framerate it attm makes zero sense to invest time into it

closed 2019-03-24 by arhi