TheSpaghettiDetective/OctoPrint-Obico #14: hook in plugin?

arhi, 2019-03-24:
from what I see plugin is one-way, sending images to the server. Can plugin fetch also the value for failure detection and have a hook that other plugins can use? or run something on the octo side? ... for e.g. I use Telegram plugin on octoprint to monitor and access my printer when I'm not at home, I would like Telegram to send me message if % go over some value I set, and I want to be able to use Telegram plugin to query the % value attm..

kennethjiang, 2019-03-25:
The bulk of the communication is from plugin to the server. But server can also send messages (currently just pause/resume/cancel commands) to plugin. Check this piece of code here: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/74c12ff353cbbe9bcfe334021210037ffb775861/octoprint_thespaghettidetective_beta/__init__.py#L216

So it's totally possible to propagate p value all the way back to plugin. The only thing is I'm not familiar (quite embarrassingly so) enough with OctoPrint plugin interface to say for sure if cross-plugin hooks are even possible. But I can totally take a stab at it when things settle a bit. Just battled a server scalability issue today. Hopefully not a lot users noticed the backend had a partial outage lol.

arhi, 2019-03-25:
no rush, this is just an idea for feature request to think about when things settle down :) .. important thing is there's mechanism to send data back and forth :)

MSkjel, 2019-05-15:
Cross-plugin support was added with the latest release, plugins can now register their own events