TheSpaghettiDetective/OctoPrint-Obico #16: Temporarily disabling

pquodling, 2019-04-04:
Is there a simple way to disable TSD for a print (short of uninstalling) - I am getting some stringing, need to print something, and am happy to mod the print with a heat gun/knife afterwards, but need it to print. Is there an easy "off" button to allow a print to happen "warts and all".

kennethjiang, 2019-04-04:
Other users made similar request which I captured at: https://github.com/TheSpaghettiDetective/TheSpaghettiDetective/issues/47

Do you think adding a toggle to "My Printer" page would be good enough for your situation too @pquodling ?

pquodling, 2019-04-05:
Sounds like a good starting point.

kennethjiang, 2019-08-22:
@pquodling I am closing this issue as we now have a toggle to temporarily disable alert for the entire print. But feel free to re-open it if you don't think that's good enough.

closed 2019-08-22 by kennethjiang