TheSpaghettiDetective/OctoPrint-Obico #17: Short freeze leads to blobs

sameSEA, 2019-04-25:
it causes short pause in printing like ~0.5 sec or less every minute (or even more frequently) on RPi 3B.  
This creates blobs when head stops on external wall.

kennethjiang, 2019-04-25:
Hmmm, that's interesting. I'm wondering why it'd cause the pause. The CPU usage should be negligible. Can you check the CPU % on the Pi with and without this plugin enabled? @sameSEA  If there is a significant jump (like >5%) on CPU % when the plugin is enabled, there is a problem.

sameSEA, 2019-04-26:
Hi,
I did few prints with this plugin enabled and disabled. The CPU load differs about ~15% in octoprint process (checked with top).
The peak without TSD was 47% at max, with TSD - 60%.

kennethjiang, 2019-04-26:
I have a feeling that somewhere in your Pi it is not optimized.

I have 2 Pi 3Bs (not B+). Both of them will have CPU at lower than 10% when printing, even if it's processing a lot of G-code such as when they are printing a small circle. 

Do you have pauses and blobs when TSD plugin is not enabled? I have a feeling that if CPU % averages at 47%, it'll shoot up to a lot higher at times and cause the printer to stutter and have blobs.

sameSEA, 2019-05-05:
I removed and disabled most plugins, that reduced cpu usage.
Then I did many prints last week with TSD off, and on. There is no stutter with TSD off, and it always there with TSD on.
I will try clean install of octopi. Though I need some other plugins installed anyway. 
Is there a way to set priorities to limit affect of TSD while printing?

kennethjiang, 2019-05-05:
I don't think there is a way to limit CPU usage from a specific plugin.

A clean install will definitely help isolate the problem. If possible, can you take these steps when you do a clean install?

1. Clean install OctoPrint. Before installing any plugins, ssh to your Pi and log the CPU usage (1-minute average) while it's not printing. This will be our base line

2. Install and configure TSD as the first plugin. then log the CPU usage again (not printing). Then start a small test print note:
    - if there are any blobs.
    - the CPU usage

3. Install other plugins one by one. Every time you install a new plugin, please note the CPU usage.

It'll be great if you can post your log here.

Thanks a lot!

kennethjiang, 2019-08-22:
Closing this issue for inactivity. @sameSEA feel free to re-open it if you still have this issue.

closed 2019-08-22 by kennethjiang