TheSpaghettiDetective/OctoPrint-Obico #18: Provide optin/optou for sentry/raven, and document what it sends

arthurlutz, 2019-05-06:
Catching errors and sending them up to sentry is awesome ! Glad to see this functionallity in the project right from the start. This shows the attention to quality and solving bugs. 

It would be nice though, for privacy conscious people to document what is being sent to the server and having a bit of documentation of what is sent in the error reports. 

You probably have to choose between an opt in or opt out approach. We could even think of having an optional sentry DSN for developer having their own sentry instance to catch bugs.

kennethjiang, 2019-05-06:
Thank you @arthurlutz for this suggestion! Good point on optin/optout.

Any chance you can create a PR for that? We can probably look at how OctoPrint does the optin/out thing and steal some code from there.

RaymondHimle, 2020-05-06:
@kennethjiang This issue is half done. Just need documentation of what is sent with Sentry

kennethjiang, 2020-05-06:
@RaymondHimle documenting sentry is a bit out of our scope. maybe we can point to this link: https://sentry.io/security/ ?

RaymondHimle, 2020-06-17:
Closing due to inactivity and optin being standard

closed 2020-06-17 by RaymondHimle