TheSpaghettiDetective/OctoPrint-Obico #19: Websocket failures after latest octoprint update

Pezmc, 2019-05-15:
Looks like octoprint might have changed the `__on_server_ws_msg__()` method?

```
2019-05-14 22:13:59,345 - octoprint_thespaghettidetective_beta.utils - ERROR - Backing off 166.857551 seconds: Failed to connect to websocket server
2019-05-14 22:14:03,095 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:14:12,187 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-14 22:14:13,114 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:14:23,127 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:14:23,137 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:14:33,159 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:14:42,259 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-14 22:14:43,181 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:14:53,210 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:03,215 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:12,329 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-14 22:15:13,228 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:23,237 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:23,247 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:33,266 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:42,403 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-14 22:15:43,285 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:15:53,306 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:03,326 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:12,473 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-14 22:16:13,340 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:23,341 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:23,351 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:33,382 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:42,543 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2019-05-14 22:16:43,401 - websocket - ERROR - error from callback <bound method MessageLoop.__on_server_ws_msg__ of <octoprint_anywhere.message_loop.MessageLoop instance at 0x697ebe90>>: __on_server_ws_msg__() takes exactly 3 arguments (2 given)
2019-05-14 22:16:46,629 - octoprint_thespaghettidetective_beta.utils - ERROR - Backing off 100.501787 seconds: Failed to connect to websocket server
```

Pezmc, 2019-05-15:
Octoprint anywhere had exactly the same problem: https://github.com/kennethjiang/OctoPrint-Anywhere/commit/eaf49169b0d0ec8602daa95abdc3531955cc4c6a

kennethjiang, 2019-05-15:
Thank you so much for reporting it! Just pushed 0.5.4 that should fix it.

closed 2019-05-15 by kennethjiang