TheSpaghettiDetective/OctoPrint-Obico #23: Setup wizard starts every time I open octopi.local

torsoreaper, 2019-06-24:
Expected behavior: After finishing the setup wizard, the wizard should not pop up when re logging into octopi.local.

Actual behavior: Every time I go to octopi.local the setup wizard pops up.  The configurations are saving correctly, I click "finish" but it doesn't seem to remember that I've done the wizard because the next time I open octopi.local it pops up the wizard.

kennethjiang, 2019-06-25:
Sorry for the delay.

We started to receive reports of this behavior after OctoPrint 1.3.11 was released. 

Unfortunately I can't reproduce the bug on my side, which makes fixing this problem trickier. Plus I suspect this is more of a bug in OctoPrint.

Can you ssh to your RPi and open `/home/pi/.octoprint/config.yaml`, and take a look at what you have in the section `server:seenWizards`?

kennethjiang, 2019-06-25:
Will appreciate help from someone who can reliably reproduce this problem and know a bit about Python debugging.

torsoreaper, 2019-06-26:
Ironically I can't reproduce it now.  I disabled the plugin until a fix was listed but when I re-enabled the plugin the problem went away.  Maybe that is a workaround solution for now?

If it still helps, under seenWizards:

```
seenWizards:
    corewizard: 3
    cura: null
    thespaghettidetective_beta: 1
    tracking: null
```

kennethjiang, 2019-06-27:
Thank you @torsoreaper for the update. My best guess is that the line
```
    thespaghettidetective_beta: 1
```
was previously missing.

This seems like a bug in OctoPrint 1.3.11. Like I mentioned, I have received quite a few similar reports since 1.3.11 was released.

Closing it for now.

closed 2019-06-27 by kennethjiang