TheSpaghettiDetective/OctoPrint-Obico #25: Use octoprint's built in logger

Prizmagnetic, 2019-06-25:
Octoprint injects a logger function for use in plugins, The Spaghetti Detective should use this feature for future debugging.
for example
`self._logger.info("some info")`
and
`self._logger.debug("some debugging info")`

see https://docs.octoprint.org/en/master/plugins/injectedproperties.html

kennethjiang, 2019-06-26:
The reason why I didn't choose to use `_logger` is when we will have to pass this var around when other classes or files need to do logging.

It's not clear to me what's the advantage of using OctoPrint logger compared to just using python's own logging mechanism: https://docs.python.org/2/library/logging.html. But I might have missed something.

Prizmagnetic, 2019-06-26:
_logger is using python's built in logger.  And currently, your logging method isn't going to octoprint.log for me

kennethjiang, 2019-06-27:
`_logger` is using python logger under the hood but it's how I'm supposed to use it (by passing it around everywhere rather than getting it out of `logging` package only when I need it).

Can you let me know which lines of logging won't go to `octoprint.log`? I can see errors from the log files users send me. Last time when I asked you to add `print` to the file and the output didn't show up was because I forgot that your OctoPrint was running as a daemon, which handles `print` differently.

Prizmagnetic, 2019-06-27:
Yes the errors work because you are raising exceptions which is an entirely different method.  Status/debug messages were not going through until I modified my client.

For example, I don't believe this line was

kennethjiang, 2019-06-27:
That log line is not showing up because the default logging level in OctoPrint is set to `INFO`. if you add a file `/home/pi/.octoprint/logging.yaml` with this content:
```
loggers:
  octoprint.plugins.thespaghettidetective_beta:
    level: DEBUG
```
You will see that line.

kennethjiang, 2020-02-08:
This issue is fixed. Now log level can be changed in OctoPrint: https://www.thespaghettidetective.com/docs/turn-on-debug-logging/

commit be744d4a (2019-06-26): closes #25
closed 2020-02-08 by kennethjiang