TheSpaghettiDetective/OctoPrint-Obico #28: TSD Plugin breaks pnotify

umamiking, 2019-08-07:
**Steps to Reproduce**

- Install the latest The Spaghetti Detective Plugin in Octoprint
- Restart the server as directed
- Change a setting that will cause a modal to appear on the right hand of the screen with a prompt. For example something that will ask if you want to go ahead with a change or not

**Observed**
- TSD breaks pnotify, a shared component that many plugins/Octoprint uses for prompts
- The user just sees the question but no buttons or way to answer. They are forced to refresh the page. They are not able to make any changes that require answering a question

**Desired**
- TSD should play nice with Octoprint and other plugins

umamiking, 2019-08-07:
Not only will I tell you how to reproduce it, I will provide you with a developer to help you fix this issue! 

STR:
- Install TSD and Octolapse
- In the Octolapse _machine_ settings, choose Manufacturer, then select a Model. 
- You'll get a pop up modal asking if you are sure you want to change your machine settings/model

O:
- Modal appears but you cannot see a OK or Cancel button. Or any buttons for that matter

D:
- Should be able to see buttons so you can make a decision.

kennethjiang, 2019-08-07:
I must have missed something but I couldn't find the manufacture/model settings for octolapse. I attached the screenshots on the pages that I think are closest. Couldn't get any modal when I changed those settings.

[image]
[image]

Can you post a screenshot @umamiking ?

FormerLurker, 2019-08-07:
@kennethjiang, he's using an rc/devel version.  I'm coming up with some code you can use to remove that pnotify line that's removing the confirm buttons globally.  Just need to test it out :)

FormerLurker, 2019-08-07:
Here you go:
```
new PNotify({
            title: "The Spaghetti Detective",
            text: "This is a test!",
            type: "error",
            hide: false,
            confirm: {
                confirm: true,
                buttons: [
                    {
                        text: "Error Details",
                        click: function (notice) {
                            notice.remove();
                        }
                    },
                    {
                        text: "Got It!",
                        click: function (notice) {
                            notice.remove();
                        }
                    },
                    {
                        text: 'Close',
                        addClass: 'remove_button',
                    }
                ]
            },
            history: {
                history: false
            },
            before_open: function(notice) {
                notice.get().find(".remove_button").remove();
            }
        });
```
Please note that I altered your popup code slightly, so you can't just cut and paste it in.  However, the only section that matters here is this:

```
{
  text: 'Close',
  addClass: 'remove_button',
}
```
(note the added class), and this:
```
before_open: function(notice) {
    notice.get().find(".remove_button").remove();
}
```
Which removes all items with the remove_button class from the current notice.

It was the only workaround I could find for this problem, and seems to work well.  I think there are better methods of doing this in newer versions of pnotify.

FormerLurker, 2019-08-07:
OH, and if you want to test against Octolapse using the version @umamiking is on, install with this link in the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/rc/devel.zip

That's an alpha version still, fyi, though it's close to an rc/devel release.

kennethjiang, 2019-08-07:
works like a charm! Thank you so much @FormerLurker for pitching in to help so quickly, and @umamiking for reporting this bug!

closed 2019-08-07 by kennethjiang

FormerLurker, 2019-08-07:
Anytime @kennethjiang!  I had similar code sitting around, so it was really no bother at all.  It's the second time I've seen this exact issue, so I was already prepared :)

I'd also like to thank @umamiking for reporting the issue and for helping to test the new version!

umamiking, 2019-08-07:
Thanks to you both for being such responsive devs. I am very motivated to help you guys improve both products as much as you can. I think they will be game-changers for the home 3D printing enthusiast.

commit 345b4a4a (2019-08-07): close #28