TheSpaghettiDetective/OctoPrint-Obico #29: Remove integrated social buttons (or load on demand)

patschi, 2019-08-07:
Due to the nature how the social media elements - like Facebook, Twitter, GitHub and the embedded YouTube video - are just, all JS/CSS files are loaded each time visting Octoprint. It doesn't play a role if you open the general or TSD-specific settings, all dependencies are loaded everytime.

See code:
[image]

See dev console (loaded just in the first second I visited Octoprint):
[image]

kennethjiang, 2019-08-07:
What's the problem with loading these javascript/css beside costing some internet data usage?

Also most the loading errors, such as doubleclick or mxpnl, in your 2nd screenshot were from other plugins I guess.

patschi, 2019-08-08:
Privacy & completely useless CPU & network usage. It loads so many libraries and increases the website loading speed each time for something I'm probably only seeing once when configuring TSD. I don't want to query Facebook's, Google's, Twitter's and GitHub's servers each time I access an internal-only webpage.

The ad servers such as doubleclick is most likely from the embedded YouTube video, as they are used by the JavaScript library loaded from YouTube.

And yes, the mxpnl is from "Mixpanel" - loaded by the Slicer plugin.

kennethjiang, 2019-08-08:
I'm interested to learn more about the privacy problems associated with loading these JS files. Can you list some specific references to why they are causing privacy concerns?

These JS files are tiny. Also browsers will cache them. So once they are loaded the first time, they won't be loaded again unless you hard-refresh the page. I don't think most users would mind loading extra 20kB once.

Isn't it a common practice to list project's social media links at the bottom of the page?

kennethjiang, 2019-08-08:
Also these JS files are loaded asynchronously. So they won't slow down the page loading.

patschi, 2019-08-08:
It's also because of efficiency. They load about 700 KB of (basically) useless stuff at the end, if you count all the things these libraries load additionally.

> ...why they are causing privacy concerns?

Because requests can be logged server-side. There are rumors (or already verified?) that Facebook tracks the user behavior when the Facebook like-thing is loaded somewhere.

Regarding privacy:
- https://www.theverge.com/2019/7/29/8934924/facebook-like-buttons-court-justice-eu-ruling-privacy-data-protection-tracking
- https://www.theregister.co.uk/2019/07/29/eu_gdpr_facebook_like_button/

> Isn't it a common practice to list project's social media links at the bottom of the page?

Yes, and I'm fine having that on a classic webpage, but not for a internal-3d-printer-management-thing. I do dislike when having 700 KB of worth overhead for 4 simple buttons and the fact that basically the half world knows when you're visting your internal OctoPrint installation. Furthermore the other fact, that all these buttons are only visible once you open the settings page (which won't be opened so often at all).

You could also simply use some static FB/Twitter/whatsoever images and link to each platform, without needing to load all that overhead of JS/CSS files. Alternatively they could be only loaded when visiting the settings menu of TSD.

But beside that, as you can read above, some pages are already removing such social media integrations from their websites (e.g. because of GDPR), or using 1-click-techniques where they are loading it on-demand once you click on them.

kennethjiang, 2019-08-08:
I assume data usage/page loading is no longer an issue (unless you really think one-time download of 700KB matters to some users).

What requests can be logged server-side? Yes FB/Google can log the fact that some browser requests their JS files, but the best they will know is a public IP of one of your ISP's routers, which is probably already known to a million of other websites including FB/Google.

I'm not an security/compliance expert myself, but my limited knowledge on HTTP/HMTL protocols makes me believe that, baring some browser vulnerabilities, there is no way for a web site to "track" a user's browsing history just by downloading a JS file from them. The fact that Google knows that you have visited an e-commerce website is because that website shares that fact with Google, not because Google's JS can somehow fetch your browsing history from your browser.

I believe GDPR is BS created by a bunch of politicians who know very little about how internet works. It happens that Gina, creator of OctoPrint, is herself a vocal GDPR criticist, which doesn't surprise as I'm sure she share similar understanding with me on how internet works. The links you sent contains no explanation on why these social widgets can cause privacy problem, except some dishonest companies share data with each other, which existed long before internet was created.

It's not I don't respect user's privacy. I explicitly asked you for permission before I posted your timelpases to TSD website, even if these time-lapses don't seem to have anything private. If I did believe these JS files would be privacy risks, I would fix them right away. But since I don't believe it, I just don't think it's worth the effort to change them to links. However I'll be happy to accept and merge a PR if anyone wants to change these social widgets into links.

kennethjiang, 2019-08-22:
I'm closing this issue assuming that it's not a problem for most users. Feel free to re-open it if anyone else feels strongly about it.

closed 2019-08-22 by kennethjiang