TheSpaghettiDetective/OctoPrint-Obico #31: Small Inversion Issue

marklair, 2019-09-10:
small issue.... when I view the live feed of TSD, the image is right side up (i may have had to invert it in octoprint)..... but when I view all of the time lapses, they are all upside down.... how to fix?

m4rcSA, 2019-10-31:
I have the camera rotated (Flip webcam horizontally & Flip webcam vertically) in octoprint, so i have the same problem with having the time lapses upside down.
The live feed looks good

RaymondHimle, 2020-02-16:
@marklair @m4rcSA Is this still an issue?

m4rcSA, 2020-03-14:
All it's ok except the telegram notification. It's the only left.
It's odd because the mail notification is correctly rotated.

kennethjiang, 2020-03-15:
Just pushed a release that should fix it for telegram as well. Please verify it and let me know if the fix works for you @m4rcSA . Thanks!

RaymondHimle, 2020-04-12:
@m4rcSA is this still an issue? @kennethjiang this likely can be closed.

marklair, 2020-04-12:
No.... This is no longer an issue

m4rcSA, 2020-04-14:
> Just pushed a release that should fix it for telegram as well. Please verify it and let me know if the fix works for you @m4rcSA . Thanks!

it's fixed! Thanks!

kennethjiang, 2020-04-14:
Awesome! Thanks folks!

closed 2020-04-14 by kennethjiang