TheSpaghettiDetective/OctoPrint-Obico #33: OctoPrint pluging  0.8.2 hight CPU ussage

luismiparra, 2019-09-28:
When I enable the pluging on octoprint Load avarage on my raspberry pi 3 goes from 0.20 avarage to 1.57.

I had not this issue with 0.8.1.

Please let me know if need any additional information.

Thanks

kennethjiang, 2019-09-28:
Do you have a USB camera? In 0.8.2 we started to stream from a USB camera in MJPEG format, which is pretty heave on CPU.

What's your Pi and USB camera resolution? When I tested it on my Pi 3B (not 3B+) with Logitech C270 at 640x480 resolution, the load average hovers around 0.25. Significantly higher than whenI have USB cam streaming disabled but seems acceptable.

kennethjiang, 2019-09-28:
We received similar report from a few other users so the problem is not isolated.

We just released version 0.8.3 to disable USB camera streaming, which is the cause of the high CPU usage.

You can get the latest version by uninstalling and re-installing TSD plugin. 

Our apologies for the problem. Thank you for being a super supportive user @luismiparra

closed 2019-09-28 by kennethjiang

luismiparra, 2019-09-29:
Hi,

First, thanks a lot for your fast respone and amazing project.

Let me add some additional details:
In my case I am running 2 instances of octoprint at my raspberry, with dual cameras on a raspberry pi3

Both cameras are "Creative Live!CAM Chat HD" and they run at VGA 10 FPS.
(I was running this config for about 6 months withouth issues)

Thanks!

kennethjiang, 2019-09-30:
Thank you so much @luismiparra for providing the extra details!