TheSpaghettiDetective/OctoPrint-Obico #34: Upgrading octoprint plugin from 0.81 to 0.83 breaks feed in personal server.

Wireheadbe, 2019-10-05:
Just upgraded the octoprint plugin from 0.81 to 0.83. Things worked well, however, the feed is now missing, while it was working before. I already restarted octoprint, but the issue remains the same. How can I help collecting logs?

Wireheadbe, 2019-10-05:
Hmm, somehow, after upgrading, when looking in Octoprint logs - I get the following: 

octoprint.plugins.thespaghettidetective_beta - ERROR - Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

However, my USB webcam works perfectly on the Raspberry.

Wireheadbe, 2019-10-05:
OK - I suppose this is due to the rollback of USB streaming? However - in Octoprint itself, my USB stream is fast, and isn't causing any lag. Why cant the plugin connect to this stream?

kennethjiang, 2019-10-05:
Since you have a USB camera, can you install a different version from this link: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/mjpeg.zip . Make sure you upgrade your server to the latest code.

Wireheadbe, 2019-10-05:
That seems to work a treat! Refresh rate is way higher as well. Load is low on the "OctoPi" - The host running the containers is fine as well. I'll start a small print and see how it's handling.

Wireheadbe, 2019-10-05:
Nice - Picture in picture with live feed + processed image "static-y" [image]

kennethjiang, 2019-10-06:
The layout of the page seems to be quite off. Have you tried that against our cloud server?

Wireheadbe, 2019-10-06:
Haven't tried it in cloud no. Note that I right-clicked on the feed and deselected picture in picture. So I can put the feed where I want. That's probably a little confusing.. :)

Wireheadbe, 2019-10-06:
I have noticed that the feed in Octoprint -> Control is now slower after installing the plugin; while the stream in TheSpaghettiDetective is fast. Is this related to changes done by the plug-in?

kennethjiang, 2019-10-07:
@Wireheadbe Yes the slower stream in OctoPrint is related to the change. the stream in OctoPrint is based on m-jpeg, while the real video streaming is based on h.264, which is incompatible with m-jpeg. We opted to have h.264 stream at full speed and slow down m-jpeg as it would otherwise put too much strain on Pi's weak CPU.

kennethjiang, 2019-10-07:
The new streaming code has been merged to the latest release. Please confirm if everything works for you now so I can close this issue.

RaymondHimle, 2020-01-17:
@kennethjiang I think this issue can be closed now.

kennethjiang, 2020-01-17:
Thanks @RaymondHimle for helping me with the house-keeping. Really appreciated!

closed 2020-01-17 by kennethjiang