TheSpaghettiDetective/OctoPrint-Obico #37: Feature Request: Honoring mjpeg-streamer configuration-options

XRyu, 2019-10-12:
It would be awesome to honor already set mjpeg-streamer settings, or enable some custom effects/filters, like greyscale, "-awb tungsten", rotation or such

kennethjiang, 2019-10-13:
Thank you @XRyu for opening this feature request. It makes a lot of sense.

I did a quick google search and found this link: https://community.octoprint.org/t/how-can-i-change-mjpg-streamer-parameters-on-octopi/203 . Is there a complete list of all the settings you can configure for camera in `octopi.txt`?

XRyu, 2019-10-13:
@kennethjiang , as i know there is this list: 
https://community.octoprint.org/t/available-mjpg-streamer-configuration-options/1106

kennethjiang, 2019-10-14:
Thanks @XRyu . It looks `octopi.txt` supports the full set of mjpeg-streamer settings. It'll be a lot of work to support them.

What options are you currently using?

kennethjiang, 2019-10-14:
Agreed that having a web interface is more user-friendly.

The problem I can see is that different cameras will probably need very different options. For instance, the white balance option supported by NoIR V2 may be very different from the ones in C910, if supported at all. It feels like a lot of work and quite error-prone to come up with a web interface that includes appropriate options for the specific camera, and to make the interface not confusing... But let me know if I missed some design options that can make it drastically simpler.

XRyu, 2019-10-14:
Hmm, i dont think so, at least when giving some parameters o users through "advanced options", like greyscale, should be the same as its only made by the processing software/encoder, not the camera.

kennethjiang, 2019-10-14:
Ok I take it back. Based on https://community.octoprint.org/t/available-mjpg-streamer-configuration-options/1106, the options between USB camera and Pi Camera are very different. But they are different in the way that USB camera barely has any options beside resolution, whereas Pi Camera has many advanced options, which may not be very difficult to implement since we don't have to deal with many different types of cameras as we would with USB cameras.

RaymondHimle, 2020-06-17:
This can now be done with compatibility mode in the TSD plugin

closed 2020-06-17 by RaymondHimle