TheSpaghettiDetective/OctoPrint-Obico #38: Streaming issue on Pi4 4gb with the last rasbian os

lele344, 2019-10-28:
Goodmorning everyone,
I have a problem with Spaghetti installed on a raspberry pi 4 with the latest version of rasbian installed.
I installed the docker with the server on another 64-bit intel pc, where the docker installation was successful (ubuntu server 64 bit).
I created the printer and connected to the plugin, it works.
the only problem is that the streaming that I see from the local server has a low refresh (1 frame for 10 seconds).
In the documentation I should have 25 fps since I have a raspberry 4 and a PI camera with IR.

The octoprint log: [here](https://drive.google.com/file/d/1J_KhzrxO90-npzMC-f5yqIb3id8-50-k/view?usp=sharing)

This is the mjpeg_streamer log:

```
Oct 21 2019 16:49:21
Copyright (c) 2012 Broadcom
version 8c7c951122d2cae572e891f04ddf287078ff2f11 (tainted) (release) (start_x)
camera: raspi
usb options: -r 640x480 -f 10
raspi options: -fps 60
/home/pi/mjpg-streamer/mjpg-streamer-experimental /home/pi
Running ./mjpg_streamer -o output_http.so -w ./www -i input_raspicam.so -fps 60
MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
 i: fps.............: 60
 i: resolution........: 640 x 480
 i: camera parameters..............:

Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 i: Starting Camera
```

Thank you for any help
Gabriele

RaymondHimle, 2019-11-17:
Try with the latest build of OctoPi as the Raspberry Pi 4 is now supported. https://octoprint.org/download/

kennethjiang, 2019-11-17:
I actually installed OctoPi 0.17 on a Pi 4 and tested it out myself with a Pi cam. It seems to work just fine.

@lele344 or anyone who is still experiencing streaming issues on Pi 4 after upgrading to 1.0.2, can you install this debugging version: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/debug.zip, then turn on debug logging: https://www.thespaghettidetective.com/docs/turn-on-debug-logging/, and post the log file here (it may be big so you may want to zip if before uploading).

kennethjiang, 2019-11-23:
closing the ticket as I have tested it on Pi 4 with OctoPi 0.17 and it worked for both USB cam and Pi cam

closed 2019-11-23 by kennethjiang

lele344, 2019-11-23:
Sorry for the delay. Unfortunately in these days I had not had the chance to try.
I printed a single piece and even from the preview I could see the camera in real time.
I also think that the update has solved the issue.
As soon as I have time to do other tests, in case of errors I warn you.
Thank you

kennethjiang, 2019-11-23:
Sounds good. Thank you @lele344 ! Glad to see it works for you now. Feel free to open an ticket if you run into any problem.