TheSpaghettiDetective/OctoPrint-Obico #42: Custom settings for camera resolution (Pi Camera only)

jmarc67, 2019-12-07:
Hello,

I have a problem with 1.0.3
You added Custom settings for camera resolution (Pi Camera only)!

I cannot see a difference between
Low, Medium, High...
My Timelapses are always 960*540 25 fps
My question is: what should I get according to the different settings? 540, 720, 1080 ?

kennethjiang, 2019-12-08:
You can see the resolutions for each level here: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/webcam_stream.py#L34

Do you see different resolutions on the real time video streaming? Remember you need to restart OctoPrint each time you change resolution.

jmarc67, 2019-12-09:
Thank you kennethjiang,

So i have a problem with that. Always 960*540 even after restart. I have the Dropbox Timelapse plugin.
In octoprint.txt my raspi is set to make stream in 1280 * 720 25 fps and since I use Spaguetti Detective my timelaps exported in my dropbox and those of Detective are all in 960*560. Could this come from the parametrage of the raspi in octoprint.txt?

However, apart from this problem of resolution, felicitation for this plugin which is very effective.

kennethjiang, 2019-12-09:
Something is definitely strange here. I don't think `octopi.txt` is a factor here as TSD plugin ignores that file. Can't think of any reason why you ended up with 960x540 resolution... :(

jmarc67, 2019-12-09:
testing:
When i disable The Spag Detective plugin (plugin manager) i have no more display in my "control" tag octoprint same if i uninstall it.
The only way i have the display back is to reinstall the plugin. Very strange...

kennethjiang, 2019-12-10:
Hah I think you found a bug! That line should have been removed when we added webcam resolution.
Can you send the email address of your TSD account to `k@thespaghettidetective.com`? I'd like to credit 100 DHs to your account for reporting this bug and helping us fix it!

jmarc67, 2019-12-10:
Hello,Oh, i'm so happy to help you to find a bug. I love your project.Thank you very much for DHs creditThe email address of my TSD account is : j-marc.67@wanadoo.frMany Thanks,Jmarc67envoyé : 10 décembre 2019 à 02:26de : Kenneth Jiang <notifications@github.com>à : TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective <OctoPrint-TheSpaghettiDetective@noreply.github.com>cc : jmarc67 <j-marc.67@wanadoo.fr>, Author <author@noreply.github.com>objet : Re: [TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective] Custom settings for camera resolution (Pi Camera only) (#42)Hah I think you found a bug! That line should have been removed when we added webcam resolution. Can you send the email address of your TSD account to k@thespaghettidetective.com? I'd like to credit 100 DHs to your account for reporting this bug and helping us fix it!

commit f12e815c (2019-12-10): closes #42
closed 2019-12-10 by kennethjiang