TheSpaghettiDetective/OctoPrint-Obico #43: Feature Request: OcotoPrint Navbar status/toggle for watching prints

barrettj, 2019-12-10:
In octoprint itself's nav-bar I'd like a status indicator/toggle for whether or not TSD is watching the print that way I can just turn off watching really quick right from OctoPrint if I don't need TSD to be watching.

kennethjiang, 2019-12-11:
I can see why this is a desirable. However, what I am concerned is when the plugin is somehow unable to talk to server (many users have unreliable internet connections for their Pis). In that case, the "watching" status on OctoPrint may go out of sync with the server, which is the source of truth. Also the toggle won't work. The code needed to handle this situation will be quite complex and error-prone.

The alternative is to use TSD page for "watching" status. Do you think it's an acceptable workaround @barrettj ?

barrettj, 2019-12-11:
My goal is more about not having to go to TSD for the times that I’m specifically not wanting TSD to be watching. 

Most of the time I’m starting a print and leaving so I like to leave TSD on.  It’ll notify my if something is wrong and I don’t have to interact with the site at all. 

But sometimes I’ll be right next to the printer and don’t want to consume hours. It’d be nice if for those times that I specifically don’t want to use TSD I didn’t have login to the site just to tell it not to watch.

A second print button in octoprint that is “print but don’t watch in TSD” would be closer to what I’m interested in (as status was more of an after thought)

kennethjiang, 2019-12-11:
Gotcha. Will incorporate it into our road map.