TheSpaghettiDetective/OctoPrint-Obico #44: Plugin causing RPi to lock up

DerekMangrum, 2019-12-27:
I have gone through the process of switching from Octoprint Anywhere to The Spaghetti Detective. After going through the complete process, I started a print. After around 1-2 hours in to the print, my RPi locked up. There was no response to the OctoPrint web interface. The printer (Ender3) still said 'Printing' on the screen, but nothing was happening.

I power-cycled both the printer and the RPi. I then started a different print. Again, after a bit more than an hour, the same thing happened again. I then uninstalled the TSD plugin and tried the first print again. It ran for all 7+ hours and succeeded. I then tried a second print, over 5 hours, and it too succeeded.

Today, I updated my RPi (apt-get update and apt-get upgrade) and then re-installed the TSD plugin. I then tried a third model to print. At an hour and a half in to the print, the RPi locked up again. I have again power-cycled the RPi and printer. I am uninstalling the plugin and trying this last print again.

I don't know if your plugin does any logging or anything. But, I would be happy to deliver any such items that might be of assistance.

Here's some specs on my RPi, if it will help.
OctoPrint version: 1.3.12
OctoPi version: 0.16.0
Linux octopi 4.19.66-v7+ #1253 SMP armv71

Details from os-release file
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian

Output from 'grep Revision /proc/cpuinfo'
Revision: a22082

kennethjiang, 2019-12-28:
Thank you for the detailed report. It does look like the plugin in somehow caused your Pi to lock up. I'm sorry about that!
Please turn on debug logging. After the Pi locks up again and you power cycle the Pi, please attached `octoprint.log` file to this issue. [Detailed instructions on how to turn on debug logging](https://www.thespaghettidetective.com/docs/turn-on-debug-logging/).
It is possible that the webcam streaming is using too much CPU, and push the power consumption of your Pi over the supply limit. Can you SSH to your Pi and run a `top` command for about 1 minute and let me know what are the most CPU-intense processes, and roughly how much CPU % they use?
Another way to test if webcam streaming is causing the issue is to disable webcam stream, but not TSD plugin and see if the problem will still occur. [https://www.thespaghettidetective.com/docs/more-about-webcam-streaming/#webcam-streaming-causing-issues]

DerekMangrum, 2019-12-28:
I have attached screencaps of the 'top' command at various times in the
run. Also in the zip file of the log file you requested.

The 'gst-launch-1.0' entry in the 'top' output is clearly growing in
resource utilization throughout the run. I don't know if this is normal, or
if it indicates a memory/resource leak. But, this looks to be the cause of
the lockup.

I have disabled Webcam Streaming in the TSD plugin and am starting another
print. I will monitor things and let you know how this print goes.

A question, with webcam streaming disabled, should I still be getting a
video feed from your service? It looks like I still do. I would have
thought that would not work.

Thank you.

Derek

DerekMangrum, 2019-12-28:
[RPi Snapshots.zip](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/files/4007344/RPi.Snapshots.zip)

Not sure the file got sent with my last entry. I think it is here, though.

DerekMangrum, 2019-12-28:
Sorry for the spam... I forgot to reboot after turning off webcam streaming. I was still seeing the gst-launch-1.0 service running and consuming resources in my next print.

I rebooted the Pi and started another print. I am no longer seeing that service running in the 'top' output. I will update further as this print proceeds/completes.

kennethjiang, 2019-12-29:
Thank you so much @DerekMangrum for the screenshots and log file. They are really helpful!

It does seem to point to `gst-launch-1.0` as the culprit.

From the log file it seems you have a Pi 3B, is that right? Also what kind of webcam do you have? I'd like to recreate a similar setup to see if I can reproduce the problem.

DerekMangrum, 2019-12-29:
Yes, that is the model of Pi I have. I am using an older MIcrosoft LifeCam
Cinema webcam.

Thank you.

Derek

kennethjiang, 2019-12-31:
Tested it for 1 day on a Pi 3B with Logitech C270 (can't find a LifeCam). Couldn't reproduce this problem... :(
For now, you can disable 25 fps video streaming while keeping TSD enabled: https://www.thespaghettidetective.com/docs/more-about-webcam-streaming/#webcam-streaming-causing-issues

We have seen more of this kind of crashes, we will see if it only happens to LifeCam...

DerekMangrum, 2019-12-31:
I disabled the streaming in the Advanced settings and the problem went
away. It looks to perform like OctoPrint Anywhere now... video, but really
low framerate. This works for me however.

I have been considering both a new RPi4 and a new camera. If I make these
changes, I will report any interesting findings on performance.

Thank you.

tjwittkop, 2020-01-01:
I am having the same issue. Recently switched from Octoprint Anywhere to The Spaghetti Detective. One print failed almost 11 hours into the print. Attempted to print the same file a couple days later and it failed about an hour into the print. 

I am running Octopi version 0.16.0, running on Raspberry Pi 3 Model B Rev 1.2, Octoprint version 1.3.12, The Spaghetti Detective version 1.0.8. My printer is an Anet A8.

Attached is my octoprint.log.

[octoprint.log](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/files/4014273/octoprint.log)

kennethjiang, 2020-01-01:
@tjwittkop What happened when your print failed? It just stopped with heaters on? Can you still load OctoPrint when it happened?

tjwittkop, 2020-01-01:
> @tjwittkop What happened when your print failed? It just stopped with heaters on? Can you still load OctoPrint when it happened?

The printer stopped with the heaters on. I was not able to connect to the pi with the web interface or Printoid. I had to power cycle the Pi and printer.

kennethjiang, 2020-01-02:
@tjwittkop Sorry for the delay. Can you also do what @DerekMangrum did by:

1. Reboot and SSH to your Pi.
2. run `top` command once every 10 minutes
3. copy or take screenshot for the output of `top` command, and post them here.

I'd like to see if your Pi crashes were caused by the same problem. Thanks a lot!

tjwittkop, 2020-01-03:
> @tjwittkop Sorry for the delay. Can you also do what @DerekMangrum did by:
>
> 1. Reboot and SSH to your Pi.
> 2. run `top` command once every 10 minutes
> 3. copy or take screenshot for the output of [...]

Attached are screenshot of the output of the 'top' command every ~10 minutes for a print that caused my Pi to lock up. While watching the processes during this print, it appears that gst-launch-1.0 used up all of the available memory on the Pi, which caused it to freeze.

I will disable premium webcam streaming like @DerekMangrum did to correct the issue for now.

[TSD.zip](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/files/4020125/TSD.zip)

kennethjiang, 2020-01-03:
Thank you so much for the info @tjwittkop . Your screenshot definitely confirms that you are having the same gstreamer memory leak.

I was testing it on exactly the same Pi (3B Rev 2) but couldn't reproduce this issue. I guess the memory leak could be trigger by the type of webcam. What webcam do you have?

tjwittkop, 2020-01-03:
@kennethjiang I am also using a [Microsoft Lifecam Cinema](https://www.amazon.com/Microsoft-H5D-00001-LifeCam-Cinema/dp/B002MCZJ78), possibly the same webcam as @DerekMangrum.

kennethjiang, 2020-01-04:
@tjwittkop @DerekMangrum My guess is the memory leak only happens to Microsoft Lifecam. I'd like to confirm my guess though. If any of you would like to buy a [supported webcam](https://www.thespaghettidetective.com/docs/webcam-feed-is-laggy/#webcams-known-to-work-with-tsd-premium-video-streaming) and test it to see if the memory leak problem goes away, I'd like to offer one-year Pro plan (1 printer) to compensate you for that. Just send email to `support@thespaghettidetective.com` to claim it.

bretcobrageek, 2020-01-07:
I haven't captured a debug log or anything, but I'm seeing a similar crashing issue Pi 3b with a logitech C270.  I will try turning off the streaming in advanced and see if the crashes go away.

kennethjiang, 2020-01-07:
That's quite concerning. Can you also run `top` command to see if gstreamer is the offender @bretcobrageek 

1. Reboot and SSH to your Pi.
2. run top command once every 10 minutes
3. copy or take screenshot for the output of top command, and post them here.

bretcobrageek, 2020-01-08:
Well, I rebooted and did the top as you asked, but for some reason TSD is indicating that the octoprint can't reach the internet (but it actually can).  So I'm not sure it's creating the problem case by sending pictures.  I'll try again later and see if I can get TSD to work again (it doesn't work easily for me).  

[image]

kennethjiang, 2020-01-08:
there is no gstreamer in your top output. Is TSD premium video streaming enabled?

What do you mean by "TSD is indicating that the octoprint can't reach the internet (but it actually can)." Is it from TSD diagnostic report?

bretcobrageek, 2020-01-08:
OK, I rebooted again and managed to get TSD up and running OK, so it's sending pictures now.  I took some top snapshots and I'm still not seeing anything exciting, but maybe you can see something there that is useful.  Let me know if you'd like anything else.

[image]

bretcobrageek, 2020-01-08:
What I meant by "TSD indicating that octoprint can't reach the internet", is when I go into the TSD settings in octoprint and hit the "test" button next to the secret number, instead of saying "you're awesome", it says "octoprint can't reach the internet".  But then I go to update manager and it is definitely able to see the updates available and download them.  Anyway, I rebooted again and now I'm "Awesome" :-).  

I'm assuming the gstreamer is gst-launch-1.0?  That is definitely showing in the above top's now that I have the video streaming.  Here's the next one:

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND
 1396 pi        20   0  460992 197048  63656 R  82.5 20.8  35:21.77 chromium-browse
  438 pi        20   0  407632  86672  13264 S  22.4  9.1  16:32.26 octoprint
 1502 pi        20   0  187176  65896   8964 S   9.9  6.9  32:49.07 gst-launch-1.0
 1521 pi        20   0  343468  92888  54396 S   9.6  9.8  30:40.72 chromium-browse
  719 pi        20   0  529976 116992  93872 R   5.9 12.3  14:31.53 chromium-browse
  115 root      20   0   29168   6572   6152 S   1.0  0.7   3:59.95 systemd-journal
  346 root      20   0   23756   3112   2164 S   0.7  0.3   2:39.68 rsyslogd

kennethjiang, 2020-01-08:
Yes `gst-launch-1.0` is the gstreamer process I talked about.

It looks like the memory usage grew from 1.2% to 6.9%, which is concerning but not nearly as bad as what other users reported. If it has the same problem, it will go all the way to 100% at which point your RPi will freeze. Can you please keep monitoring it to see if it's the case? Thanks a lot!

bretcobrageek, 2020-01-08:
I will try to redo this experiment in a couple days when I can be home for a few hours.  I stopped monitoring to get some sleep and when I got up it was locked up and the pi is not responding.  I'll attempt again Thursday, when I think I should be able to monitor for a couple hours.  Thanks.

talandar, 2020-01-08:
I'm also experiencing this problem.  I have a Logitech C310 as my camera.  I disabled the plugin to get some prints to actually work, but I can run some tests tonight to see if gst-launch is the culprit for me as well.

kennethjiang, 2020-01-08:
Yes your help will be really appreciated @bretcobrageek @talandar . Initially I thought it was caused by MS LifeCam, but @DerekMangrum has confirmed that he still has the same problem with a Logitech cam. It looks like more and more users are running into this issue, which is serious as it'll freeze the entire pi, and I can't think of a good way to fix or even detect it.

Hopefully we can put our heads together and figure out the failure pattern and eventually the root cause of it. Thanks a lot!

talandar, 2020-01-09:
I ran a script that did a top every minute and put it into a csv file while I waited for the print to fail.  I haven't spent a lot of time analyzing it, but gst-launch1.0 does seem to leak memory - it increases in a pretty linear manner for the duration of the print, then the pi freezes. [this pastebin](https://pastebin.com/f4ypjMEm) is a csv from top's output for that process.  

These data points SHOULD be one minute apart, but the timestamps are kind of interesting.

For each of the output files, the time between the files increased fairly regularly at one minute intervals, as expected, till near the end.  The time delta between the last few files are 1.5 minutes, then 3 minutes, implying that my very simple ```sleep 60``` wasn't getting enough cpu time to wake up on time.

DerekMangrum, 2020-01-09:
So, I imaged a new OctoPi instance with the current version (0.17.0),
installed my plugins again, and ran another print. Unfortunately, same
issue... gst-launch-1.0 mem usage grows until RPi freezes.

Derek

kennethjiang, 2020-01-09:
@talandar you are shell script expert! Do you mind sharing your script here so that we can all run it when we need to test it?

Also I think the 3 columns in your output, from left to right, are virtual memory, residential memory, and shared memory respectively?

bretcobrageek, 2020-01-09:
Yes, if you share your script, I'll run it tonight and see if that helps.  Kenneth, would you prefer to update to the latest release 1.1.0, or stick with the prior release that we were using (I think 1.0.8)?

kennethjiang, 2020-01-09:
@bretcobrageek 1.1.0 doesn't have any update on video streaming. it only added remote g-code upload and printing capability. So yes please upgrade.

talandar, 2020-01-09:
In retrospect, I should have put the script here when I posted my data.  Regardless:
```
echo "start!"

while true; do
        sleep 60
        echo "output"
        top -b -n 10 | sed -n '8, 19{s/^ *//;s/ *$//;s/  */,/gp;};19q' &> "$(date +%Y-%m-%d_%H-%M-%S.txt)"
done
```
This will create a file from the first ten entries in top.  I recommend running it in a folder by itself, just for ease of cleanup.  It will also say "output" every minute while it is working.

The columns in the output are exactly the columns from top, but in csv format:
```
PID, USER, PR, NI, VIRT, RES, SHR, S, %CPU, %MEM, TIME+, COMMAND
```
So virtual memory is the 5th column, Residential memory is 6th.  Memory used is the 10th column.

kennethjiang, 2020-01-09:
Awesome! Thank you so much @talandar for sharing the script and the detailed explanation!

bretcobrageek, 2020-01-10:
OK, I rebooted everything and have been running this script for hours.  But for some reason the gst-launch memory is still at 1.2.  It has never budged.  The printer is on and connected, but it's not actually printing anything currently.  Do the rest of you need to be printing something before the memory leak kicks in?  Maybe there is something else Iv'e done that is avoiding the issue?

kennethjiang, 2020-01-10:
That may be it! I have been running test without printing assuming that whether or not it's printing doesn't matter! But it may! Because when it's printing TSD plugin will read jpg from gst's tcpsink!

That's a really good observation! Will run the test again while printing!

tjwittkop, 2020-01-10:
Yes, the failure only happens while printing. The printer can sit for days without printing without the failure.

bretcobrageek, 2020-01-13:
OK, so I ran for a couple hours while I could be home to collect data and used talandar's script (modified to 2 minutes) to capture a bunch of outputs. It shows a very linear increase in memory usage.  It maxes out at 57.5 before crashing.  There was one point where it dropped from 4.8 to 4.0 and that was when I had to stop a print (which was printing from SD) and restart it a couple minutes later.  

At this point I have turned off TSD so it will stop crashing my pi.  Let me know if you'd like other data and I'll see what I can do.  I may switch back to octoprint anywhere if that still works until you can figure this one out. Thanks so much for working on all these great tools and sharing them with the community!
[tops.xlsx](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/files/4052265/tops.xlsx)

kennethjiang, 2020-01-14:
Thank you so much @bretcobrageek for helping us test it and collecting data! Please feel free to switch back to OctoPrint Anywhere for now. I may plan to keep OctPrint Anywhere for longer than expected just for subscribers. If you are already a subscriber you can stay in there for longer. Otherwise. please give me your OctoPrint Anywhere email and I'll be happy to give you a complimentary Pro account in OctoPrint Anywhere.

bretcobrageek, 2020-01-14:
Thanks Kenneth (@kennethjiang ).  I am not a subscriber for OPAnywhere but I would be glad to accept your kind offer.  My account email is the same as my username here, bretcobrageek and it is at gmail.  I'll switch back to Anywhere in a few days then.  Thanks!

matrimcauthon11, 2020-01-15:
Hi Guys, just wanted to let you know I am seeing the same freezing issue on my 4 I think.

kennethjiang, 2020-01-22:
We have a testing version available that hopefully addresses the streaming memory leak problem. Please help us test it! To do so:

1. Install the testing version using this URL: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/recycle_gst.zip
[image]

2. Make sure "disable premium webcam streaming" is unchecked.
[image]

3. Start a print and observe if you still have the same problem as you did before.

kennethjiang, 2020-01-22:
@DerekMangrum @tjwittkop @bretcobrageek @talandar @tjwittkop @matrimcauthon11 Will really appreciate it if you can help us test it!

DerekMangrum, 2020-01-22:
I have sprint running that will end tomorrow evening. I will test right
after it finishes.

Thank you!

Derek

talandar, 2020-01-22:
I've installed the test version and it's watching a print right now.  Looking at top, the memory is creeping up already, but we'll see if it's a problem. 

Pi model: 3B+ rev 1.3
Webcam is Logitech C310
OctoPi is 0.15.1
Octoprint 1.3.12

kennethjiang, 2020-01-22:
Thanks @talandar . This "fix" is really just a bandaid that will kill and restart gst when the memory climbs too high. So I'm not surprised to see gst's memory usage still goes up.

Have tried many different ways to figure out the root cause, including even trying to replace gst with ffmpeg, but it turned out to be really hard... So had to put in this bandaid as it seems more and more users are impacted by this issue.

talandar, 2020-01-22:
Unfortunately, I still had the pi freeze on me.

https://pastebin.com/yQNaSpTQ
that's the pastebin with the results of running my script from above (and grepping for gst-launch).  Seems unchanged. :(

kennethjiang, 2020-01-22:
It looks like for some reason the script didn't kick in.

@talandar  can you reboot Pi, ssh to it, and run:
`ps -ef | grep kill_leaked_gst`

It'll be even better if you can email me at `k@thespaghettidetective.com` to set up a video call to do this debugging in real time. It's been a little like shooting in the dark as I can't reproduce this issue myself.

kennethjiang, 2020-01-22:
You can also join us at our discord forum: https://discordapp.com/invite/NcZkQfj , which is very active and helpful to everyone in there.

talandar, 2020-01-22:
Just checked, and I had three instances of those scripts running.  Interestingly, when I ran it again to paste in here, there were only two.  After another reboot, back to three.  Not sure what's going on there.

I'll email you about troubleshooting.

kennethjiang, 2020-01-22:
Found a couple of issues in my script. Just pushed another commit. Please repeat the installation process to get the latest code. @talandar if you get a chance you can test the latest version before we call this week. I am 75% confident the latest commit will fix it.

talandar, 2020-01-22:
running it now.  Should be able to report back with results within a couple hours.

kennethjiang, 2020-01-22:
awesome! Thank you so much!

talandar, 2020-01-22:
unfortunately, same crash as before.  I'm going to do some bash fiddling to see if I can figure out why.  The kill-script is comparing some value to 200000, which value are we expecting that to be?  The first version of the script looked like it was looking at resident memory, the second at virtual.  

I'm doing some debugging on the script, trying to see what's happening.

kennethjiang, 2020-01-22:
the intention is to compare resident memory against 200k (200MB).

On my pi. the first version pulled the shared memory column. The latest commit pulled the resident memory. You the shell script master so feel free to suggest some more robust way to achieve that!

talandar, 2020-01-23:
Created #47 to address this.  I've had a print running and detection going for much longer than before, and the gst-launch process has been killed and restarted successfully twice now.  I would definitely advise more testing both from people who did and did not experience the issue before.

kennethjiang, 2020-01-23:
Merged with some minor changes. Can you help me test again to see if it works for you (I can't really test it as my gst memory usage barely moves)? If it works for you, I'll merge it to master and cut a release as this seems to be a relatively safe change that doesn't require extensive testing.

talandar, 2020-01-23:
starting up a longer print this morning with it running the test version.  I've verified the scripts are the newest versions.  I'll report back after work.

kennethjiang, 2020-01-23:
Perfect thanks again @talandar

DroneMang, 2020-01-23:
I also have had my printer locking up.  I was watching the memory fill up in the Resource Monitor Plugin.  Took me a while to figure out what was going on.  I have a logitech camera.  Does this issue occur if using a RPi camera with the ribbon cable?  Should I disable the streaming, or just go back to octoprint anywhere for now?

kennethjiang, 2020-01-23:
@DroneMang This memory leak issue happens only to USB cameras. Pi Camera with ribbon cable is fine.

You can go back to OctoPrint Anywhere but the Free plan will end of life soon. The Pro plan unfortunately doesn't support 25 fps streaming for USB camera (because it's hard!)

If you want, you can try [this testing version](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/44#issuecomment-576967554) to see if it fixes the memory leak problem.

[23 more replies]

closed 2020-01-28 by kennethjiang