TheSpaghettiDetective/OctoPrint-Obico #45: No more Time-lapse

jmarc67, 2020-01-10:
Hello, i have a problem. 
All is working  perfectly except I have nothing here: https://app.thespaghettidetective.com/prints/ no more time-laps. 
What can i do ? I was working before but now since 1.08 nothing. Empty prints page

If you have any idea.

Thanks

RaymondHimle, 2020-01-11:
I see this as well on my self hosted server.

jmarc67, 2020-01-11:
@RaymondHimle: Yes it' very strange. I don't know why this is like that now. All was ok before.

kennethjiang, 2020-01-11:
@jmarc67 I thought it was something in your plugin set up but it turned out to be a bug our code  introduced Thursday. I'm pushing it to prod right now and will re-generate all the missing time-lapses

jmarc67, 2020-01-11:
@kennethjiang Thanks, good news.

RaymondHimle, 2020-01-14:
This issue can be closed. Issue has been fixed in production and private-servers.

closed 2020-01-14 by jmarc67