TheSpaghettiDetective/OctoPrint-Obico #46: feature request - method to retrigger detective if left off

Dloranger, 2020-01-14:
I turned off the detective due to a print it really didn't like (but was actually fine), being a forgetful human, I forgot to re-enable the detective through the webpage and thankfully caught a failure myself and aborted the print.

What I would like is a way to automatically re-enable the detective for each print.

I can see 2 approaches
1) just have it automatically enable when a new print is started
2) give a g-code hook to trigger it such that it could be built into the start-up scripts in the slicer.  I have this already for my LED bar which uses "M150 W" to enable the lights and "M150" to disable them.  the printer just ignores this command since it doesn't know what M150 means.  Works flawlessly

kennethjiang, 2020-01-15:
Really good suggestions @Dloranger .

2 seems to be little better than 1 (but let me know if you see a big pro in 2 but not 1).

The tricky thing is how will be the optimal user experience for users who want to turn off watching for multiple prints in a roll. It probably has been more common now that DHs are no longer free. Some users may want to turn off watching so that they can save the DHs for "risky" prints. In this case, they'll be pissed off if we automatically turn on watching at the beginning of a new print and they don't notice.

We talked about having both A) turn off "watching" until I turn it back on, and B) turn off "watching" for current print" toggles next to each other. However no matter how we design the interface and the wording, they still look very confusing to many users.

Ideas?

Dloranger, 2020-01-15:
I think this is where option 2 really works best, you can do a simple edit to the gcode (comment or uncomment as you please with a basic startup script entry) and every time you print the same file it will use the same settings so if you are running a batch job there will not be any user intervention needed.

Default condition with a missing script file _could_ be to turn on the detective for the folks that are just getting started and have yet to tweak their startup scripts.

When uploading a file via the web, it could look for the code and if it doesn't find it, it could prompt the user to decide and then inject the code into the file.

As for the UI radio buttons, I could see this being tough no matter which way you go, and in my workflow, it would not matter which way you go, especially since I never launch a print from the UI. 

I launch all my prints from within cura (octoprint plugin) or directly from octoprint if I am doing a reprint.  Having something on the web UI would ultimately be meaningless for my workflows as I would not see it unless octoprint UI threw up a blocking message somehow that I had to address the question before the print could run.  Not sure how the Cura plugin would handle this as well.

kennethjiang, 2020-01-17:
Sorry for the delay. A lot of activities in discord and email so github is now neglected... :)

I can see your perspective as a power user. However I have a feeling that probably only a small percentage of the users are comfortable with controlling things not directly related to the model geometry through gcode.... But I can ask for it when I talk to other users.

One question, if you had a gcode in your cura startup scrip to enable watching, and you are slicing for the next model that you don't want it to be watched. What would you do to remind yourself taking that from the final gcode? It's embarrassing but I have to admit that every time I slice for something I feel a bit overwhelmed by all the knobs to tune. For instance, it has happened multiple times that I forgot to adjust the fan settings appropriately when I print in ABS... So I feel that having to worry about yet another line of gcode doesn't seem to fare well with my aging brain. ;)

Dloranger, 2020-01-18:
I thankfully dont fiddle with my levers too much, mostly I have those all
setup in my filament setting or just leave them static.

I recon if I had to toggle between them frequently I would possibly clone
my printer (in cura) with one as watched and one as unwatched.

If I did this I could then also prefix the filename with watched/unwatched
or some other trigger to remind myself when printing what to expect.

Might come up with some other technique, just what comes to mind off the
top of my head