TheSpaghettiDetective/OctoPrint-Obico #50: Add camera control mechanism

RaymondHimle, 2020-02-03:
Add way to allow other plugins and user to edit camera configuration from OctoPrint. https://github.com/FormerLurker/Octolapse/issues/426

kennethjiang, 2020-02-04:
Is this different from #37?

RaymondHimle, 2020-02-04:
More of a different means. Rather than trying to implement mjpeg-streamer options create a set of options that can be applied to either what TSD uses or to mjpeg. More of an API for controlling camera settings that any streamer could take in if supported by said streamer. Also allowing other plugins to change these options without being aware of the actual streamer used.

kennethjiang, 2020-02-04:
Do you envision Octolapse calling TSD api to adjust camera settings? 

My hesitation here is that using H.264 entails a lot more constraints on streaming than using MJPEG (as in mjpeg-streamer). The biggest issue is to have both H.264 for streaming and MJPEG for snapshotting, without blowing up Pi's CPU.

RaymondHimle, 2020-02-04:
I think we should do some testing to see the affect on the Pi CPU. It might be good to get FormerLurker's input to see what would be possible in terms of getting the control octolapse needs for the camera in TSD

FormerLurker, 2020-02-04:
I've been thinking about other ways to control the camera stream/snapshot url picture, but haven't found any good multi-platform solutions as of yet.  Ideally there would be an API similar (better yet, identical) to mjpeg-streamer's control.htm that provides a list of inputs, options, min/max and step.  I'm not sure if you know how mjpeg-streamer handles camera image controls or not, but if you navigate to http://{ip_of_pi}/webcam/input.json, you can see how the controls are reported.  You can ignore the current resolution stuff, since as far as I know this functionality was not implemented.

Once I have the control list I can render all of the available camera controls.  Changing the settings is a simple matter of constructing the appropriate URL with the correct parameters and sending one request per contorl to the mjpeg-streamer's control.htm page.

The worst case scenario for me would be to have to write separate routines for many different servers.  I've also been thinking of splitting the camera controls from Octolapse eventually, which would give me a lot more flexibility.

@kennethjiang, I need to take a better look at your custom streaming code to get a better idea regarding what is even possible when streaming with H.264.  You are definitely the expert here.

FormerLurker, 2020-02-04:
FYI, this command lists most of what would be needed, but not the menu information.  

```
v4l2-ctl --all
```

I would need to do more research to see if the menus can be retrieved, and if there is an equivalent command for windows/mac.

kennethjiang, 2020-02-04:
Thanks @FormerLurker  for your insights. The current streaming code is actually quite ugly. Patches over patches to make things work for like 95% of the users. But will be happy to walk you through the code in case you are interested.

FormerLurker, 2020-02-04:
@kennethjiang, I'm quite happy to assist/cooperate in any way I can.

> The current streaming code is actually quite ugly.

To prevent ocular damage, you may not want to look into the Octolapse code too much, lol!

RaymondHimle, 2020-04-25:
Placing this here for future contemplation https://github.com/Ernie3/v4l2-ctl-rest-api

kennethjiang, 2020-07-17:
@RaymondHimle Do you think the "compatibility mode" is an acceptable workaround for this problem?

RaymondHimle, 2020-07-17:
In fact I do. :)

closed 2020-07-17 by RaymondHimle