TheSpaghettiDetective/OctoPrint-Obico #56: Pause detection on filament runout

RaymondHimle, 2020-04-06:
Pause spaghetti detection on filament runout.

kennethjiang, 2020-04-08:
We definitely want to do something about filament runout, such as pausing the detection, as well as alerting user.

What I don't know is if there is some standard G-code or OctoPrint API so that we don't have to handle it based on different make/model/plugin being used.

RaymondHimle, 2020-04-09:
There isn't really standard gcode for this at all. It might be best to figure out how the most popular printers and firmware alert that the filament it out

kennethjiang, 2020-04-09:
That'd be quite tricky. Maybe we should build it as "mini-plugin" framework so that other developers can plug in the ways that will work for the printers they own?

RaymondHimle, 2020-04-09:
Might be worth that. Have some sort of common event that plugins can call to alert of filament runout. But how do we detect when it resumes printing? Because the pause on runout on Marlin just pauses the printer not octoprint.

midnightsmith89, 2022-01-06:
M600 works with just about every printer that uses it in firmware.  Creality, Prusa, Monoprice, etc