TheSpaghettiDetective/OctoPrint-Obico #60: Support x86 Linux for 25FPS Streaming

RaymondHimle, 2020-04-24:
Seeing the number of x86 platforms used to run Octoprint increase. Would be worth compiling and adding secondary bins for Janus and GST for x86 Linux.

RaymondHimle, 2020-04-29:
Expand to include 64bit ARM architectures.

kennethjiang, 2020-04-29:
Adding bins are quite straightforward. What is holding me back are:

1. Testing. GST seems to be very buggy. To support x86 and ARM64 means testing it on a myriad of combinations of hardware and OS/lib versions. Don't think this is realistic for us unless we have a way to accomplish it in a more reliable way.

2. (More of a minor concern) I tried pretty hard to cut down the size of the final package. Right now it takes a couple minutes to install/upgrade TSD plugin on a Pi, which I think is close to the boundary before some users start to think the installation is broken and kill it. Since the bulk of the current package size comes from these binaries - supporting other bin formats means tripling the size of the final package. I think there is a way to work around this problem but it'll take some extra work.

Suggestions/thoughts?

Tr0llmops, 2020-05-31:
Several

1. Why don't add it as an experimental (opt-in) feature? Worst case: It won't work, same as now. But in many cases it probably will work, and this way you might gain some new users and feedback on different environments. Alternatively, you could provide some hints on how compile those binaries yourself, so that tinkerers like me who are willing to take the risk know where to start.

2. At least improve the basic streaming fallback (for pro users), e.g. to 1 FPS or a user-adjustable value. That way it won't hurt so much when 25FPS doesn't work.

3. Please mention this restriction at the appropriate places. This isn't just a cause of "laggy streaming", this is a system requirement for your Number One premium feature. If it wasn't for the free trial, I would be severely pissed right now.

kennethjiang, 2020-06-01:
@Tr0llmops All you said make sense. The only problem is the constraints on our engineering resources. :( The tasks (including this one) in our backlog is about 3-4 times of what we could accomplish, so we have to prioritize the tasks that would maximize the benefits to the entire TSD user community.

Having said that, we welcome contribution from anyone since TSD is open source. If you think this feature is important, you can pitch in a PR. If your contribution is significant enough, which should be true if you manage to solve this issue, you will earn yourself a place in the team section on https://www.thespaghettidetective.com/