TheSpaghettiDetective/OctoPrint-Obico #61: Feature Request: Add support for different camera feeds

purist69, 2020-05-23:
I have two camera's in my setup, an RPi camera and a USB camera.  By default the RPi camera is in fact the default for the system and is used to record TheSpaghettiDetective feeds.  In the MultiCam Plugin (and the default video configuration) I have defined my USB camera as the default feed, but TheSpaghettiDetective looks to use the system default camera vs. the Octoprint defined default camera.  It would be great if there was a way to select which feed to use in TheSpaghettiDetective.

kennethjiang, 2020-05-24:
Thank you for submitting this issue.

I need to experiment with MultiCam plugin to fully understand it. Will get back to you on this issue later.

purist69, 2020-05-24:
To be specific, in 1.4.0 OctoPrint uses services to start the webcam feed.  /etc/systemd/system/webcamd.service starts /root/bin/webcamd.  Within /root/bin/webcamd is a reference to /boot/octopi.txt.  you can either update your camera settings in the referenced octopi.txt or directly in webcamd.  To get the second feed I created a /etc/systemd/system/webcam2d.service, and registered it, added a /root/bin/webcam2d and modified the cam settings as I wanted and created a /boot/octopi-cam2.txt, referenced it in webcam2d, but left it without any update values.  You may also need to specify the video feed in webcam2d.  You may have to hard code the video source.  you can check what camera you have and the video source it's using by typing:

v4l2-ctl --list-devices

Within your webcam2d config modify the usb options by appending the video source:

camera_usb_options="-r 1920x1080 -f 30 -d /dev/video1"

It's important to note that TheSpaghettiDetective disables the second video feed (in my case the USB feed) when in compatibility mode.

Hope this help!

-Stuart

kennethjiang, 2020-05-24:
Thank you for the details!

TSD would disable `webcamd` service in "advanced mode" streaming. However in compatibility mode it would leave `webcamd`  enabled. If you found your `webcamd` disabled in compatibility mode, it is a bug.

tyeth, 2021-12-16:
This majorly interests me, I wish to use two USB cameras on separate USB buses (3.0 and 2.0) on an 8gb Raspberry Pi 4. I also was considering attempting to use an on board camera aka picam, and considering an IPcam (4k nanny cam). 

I have multicam and octolapse setup to support the nozzle cam while TSD watches the enclosure cam, but ideally TSD would watch both. 

Just thinking aloud I guess it would be possible to run a cloned and renamed octoprint plugin, or keep posting every 30seconds the 2nd camera as video recordings to the TSD website. This should work as you can upload time-lapse to TSD and rely on real-time detection just for one camera, while possibly benefiting from near real-time on the second camera - if you were to check the result while uploading the next segment you could save polling for a result.

beccsjb, 2022-01-16:
I too would love Multiple cameras available in TSD. It seems the normal now that people run an overview camera and a Nozzle Camera! No need to have them running through TSD at the same time, youd only want to view one at any time. 

I Agree that you should be able to define the camera that TSD uses to analyse a Spaghetti Monster

kennethjiang, 2022-01-17:
@beccsjb Can you let me know how you set up your cameras? Probably a link to the tutorial/video you followed?

beccsjb, 2022-03-09:
Sorry @kennethjiang i didnt see your reply there!
The link below is the link i followed to get multiple cameras working;
https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues/61

kennethjiang, 2022-03-09:
@beccsjb No problem. But I think you posted the wrong link. Can you repost it?

beccsjb, 2022-03-09:
@kennethjiang
https://community.octoprint.org/t/setting-up-multiple-webcams-in-octopi-the-right-way/32669