TheSpaghettiDetective/OctoPrint-Obico #62: 1.2.2 Breaks video

purist69, 2020-05-24:
Since installing 1.2.2 this morning my dual feed video has stopped working, specifically the USB feed is no longer working and the raspi camera quality is poor, similar to compatibility mode.

I DID try compatability mode this morning for the free 100 hours of free Detective Hours.

After rebooting a number of times I got it working again, but had to completely remove 1.2.2.

purist69, 2020-05-24:
I've not been able to reinstall it either, so I can't test changes.

purist69, 2020-05-24:
I was able to get it reinstalled, same issue though.  to recover properly, I need to uninstall, then reboot, then power cycle and then everything re initializes properly.

kennethjiang, 2020-05-25:
Can you turn on debug logging and send the log file to `k@thespaghettidetective.com`?

1. Turn on debug logging
2. Restart OctoPrint.
3. View TSD page and wait for a couple of minutes.
4. Download "octoprint.log" file and send it to me.

Find the detailed instructions at: https://www.thespaghettidetective.com/docs/turn-on-debug-logging/

purist69, 2020-05-25:
sorry to say I'm unable to, I've taken the raspi cam out of my configuration and have repurposed it, partially to get this working, but I also needed the cam for another purpose.

kennethjiang, 2020-05-30:
Most of the bugs in 1.2.2 have been fixed in 1.2.4. I'm closing it for now. Feel free to re-open it if you still experience problems.

closed 2020-05-30 by kennethjiang