TheSpaghettiDetective/OctoPrint-Obico #63: Python 3 compatibilty not reported in OctoPrint plugin repo

maddogjt, 2020-05-26:
The OctoPrint plugin repo still reports TSD as not having Python 3 compatibilty, despite support being added in release 1.1.5.  Python3 users cannot install via plugin repo.

Github stats
Latest Release: 1.2.3
Releases: 35
Last update: 25 May 2020
Stars: 54
Open Issues: 21

Compatibility Information
OctoPrint: 1.3.2+
Operating Systems: Linux, Windows, Mac OS X, FreeBSD
Python: >=2.7,<3

kennethjiang, 2020-05-28:
Thanks @maddogjt . Will look into it.

kennethjiang, 2020-07-17:
Sorry for taking so long to get to this issue.

I couldn't reproduce this problem. What I did was to following this guide https://docs.octoprint.org/en/master/plugins/python3_migration.html#how-to-get-a-python-3-virtual-environment-with-octoprint to set up a python 3 virtualenv and start OctoPrint in it. I was able to install TSD plugin from the plugin manager.

Also the plugin page here shows that it's python 3 compatible: https://plugins.octoprint.org/plugins/thespaghettidetective/

How did you set up your OctoPrint to run in Python 3?

maddogjt, 2020-07-17:
It looks like the manifest got updated a few weeks ago to properly reflect python3 compatibility:

https://github.com/OctoPrint/plugins.octoprint.org/blame/17c68713f6e347a193410913139f870f650446e0/plugins.json#L6652

The change that triggered this update appears to have been a fix from foosel related to pulling metadata from plugin github repositories, so it's possible that's was blocking the compatibility getting updated.

closed 2020-07-17 by maddogjt