TheSpaghettiDetective/OctoPrint-Obico #64: Own Server dialog missing on current plugin.

richardhughes260, 2020-06-13:
Own Server dialog missing on current plugin

The dialog for the checkbox and server address are no longer under the advanced tab.

kennethjiang, 2020-06-13:
Can you please hard-refresh the OctoPrint page to see if the setting will show up? If not, please look for javascript errors and let me know if you see any: https://documentation.concrete5.org/tutorials/how-open-browser-console-view-errors. 

Also please post a screenshot here. Thanks!

RaymondHimle, 2020-06-17:
@richardhughes260 any updates on this issue?

richardhughes260, 2020-06-17:
SRY! Missed the first note. I installed an older plugin, it worked, then I upgraded, that worked.
Maybe I needed a system restart before I began the process. All is working fine now.

RaymondHimle, 2020-06-17:
Closing resolved issue

closed 2020-06-17 by RaymondHimle

RaymondHimle, 2020-06-17:
@richardhughes260 Do you have the themify plugin install by chance?

richardhughes260, 2020-06-17:
no idea what that is.
Might remove Slic3r, have not used it.

[image: image.png]