TheSpaghettiDetective/OctoPrint-Obico #66: Plugin cannot install on Octopint 1.4.0 on Pi 4 with HQ camera

Lefuneste83, 2020-06-24:
On a fresh Octoprint install 1.4.0 with the latest Raspberry Camera HQ installed and running (immage in Octoprint).

--> /boot/octopi.txt

camera="raspi"
camera_raspi_options="-x 2028 -y 1520 -fps 25"

I cannot install the Spaghetti detector plugin. It crashes when installing the picamera 1.13 library.

Building wheel for picamera (setup.py): started
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-OUvLG0/picamera/setup.py'"'"'; __file__='"'"'/tmp/pip-install-OUvLG0/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-DS0nhb
cwd: /tmp/pip-install-OUvLG0/picamera/
Complete output (54 lines):
running bdist_wheel
running build
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/picamera
copying picamera/mmalobj.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/array.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/frames.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/camera.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/renderers.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/__init__.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/bcm_host.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/mmal.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/encoders.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/display.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/color.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/streams.py -> build/lib.linux-armv7l-2.7/picamera
copying picamera/exc.py -> build/lib.linux-armv7l-2.7/picamera
running egg_info
writing requirements to picamera.egg-info/requires.txt
writing picamera.egg-info/PKG-INFO
writing top-level names to picamera.egg-info/top_level.txt
writing dependency_links to picamera.egg-info/dependency_links.txt
reading manifest file 'picamera.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
no previously-included directories found matching 'debian'
no previously-included directories found matching 'docs'
writing manifest file 'picamera.egg-info/SOURCES.txt'
installing to build/bdist.linux-armv7l/wheel
running install
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-OUvLG0/picamera/setup.py", line 145, in <module>
main()
File "/tmp/pip-install-OUvLG0/picamera/setup.py", line 140, in main
cmdclass             = {'install': CustomInstallCommand},
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
return distutils.core.setup(**attrs)
File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
dist.run_commands()
File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
self.run_command(cmd)
File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
cmd_obj.run()
File "/home/pi/oprint/local/lib/python2.7/site-packages/wheel/bdist_wheel.py", line 228, in run
self.run_command('install')
File "/usr/lib/python2.7/distutils/cmd.py", line 326, in run_command
self.distribution.run_command(command)
File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
cmd_obj.run()
File "/tmp/pip-install-OUvLG0/picamera/setup.py", line 109, in run
raise ValueError('This system does not appear to be a Raspberry Pi')
ValueError: This system does not appear to be a Raspberry Pi
----------------------------------------
ERROR: Failed building wheel for picamera
Building wheel for picamera (setup.py): finished with status 'error'
Running setup.py clean for picamera
Successfully built Access-Anywhere-The-Spaghetti-Detective
Failed to build picamera
Installing collected packages: picamera, Access-Anywhere-The-Spaghetti-Detective
Running setup.py install for picamera: started
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-OUvLG0/picamera/setup.py'"'"'; __file__='"'"'/tmp/pip-install-OUvLG0/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-MZl_ex/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/picamera
cwd: /tmp/pip-install-OUvLG0/picamera/
Complete output (18 lines):
running install
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-install-OUvLG0/picamera/setup.py", line 145, in <module>
main()
File "/tmp/pip-install-OUvLG0/picamera/setup.py", line 140, in main
cmdclass             = {'install': CustomInstallCommand},
File "/home/pi/oprint/local/lib/python2.7/site-packages/setuptools/__init__.py", line 145, in setup
return distutils.core.setup(**attrs)
File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
dist.run_commands()
File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
self.run_command(cmd)
File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
cmd_obj.run()
File "/tmp/pip-install-OUvLG0/picamera/setup.py", line 109, in run
raise ValueError('This system does not appear to be a Raspberry Pi')
ValueError: This system does not appear to be a Raspberry Pi
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-OUvLG0/picamera/setup.py'"'"'; __file__='"'"'/tmp/pip-install-OUvLG0/picamera/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-MZl_ex/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python2.7/picamera Check the logs for full command output.
Running setup.py install for picamera: finished with status 'error'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Lefuneste83, 2020-06-24:
After searching for this specific codes, I amanged to first install the picamera library by doing the following :

sudo -s
export READTHEDOCS=True
/home/pi/oprint/bin/python2 -m pip install picamera

Once this succeeded, I could go back to Octoprint and install the plugin without error.

kennethjiang, 2020-06-25:
Thank you for doing the research yourself @Lefuneste83 .

My gut says this is not related to the fact this is an HQ camera. I think I have seen similar error even before HQ camera came out.

In any case, I'll leave this issue open for now just in case other users have run into it.

ksummers92, 2020-07-15:
Thanks for posting this in the issues with a solution @Lefuneste83 and thank you @kennethjiang for leaving it open.

@Lefuneste83 you're solution fixed my problem which was a similar issue.

To add to what @kennethjiang said about it not being related to the HQ camera, I was having this issue with a Pi 4 and an arducam. The commonality here appears to be the Pi 4.

Thanks again.

RaymondHimle, 2020-07-22:
Seems to be an issue with PiCamera itself https://github.com/waveform80/picamera/issues/630

kennethjiang, 2020-07-28:
@ksummers92 @Lefuneste83 Can you guys run `cat /proc/cpuinfo` and post the output here? We have got a lot more users reporting this problem recently and we suspect this is because of a new Pi 4 revision that is breaking `picamera`. Thank you so much!

ksummers92, 2020-07-28:
Here is the output @kennethjiang.

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 135.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03111
Serial		: 10000000d271cd93
Model		: Raspberry Pi 4 Model B Rev 1.1

kennethjiang, 2020-07-28:
Thank you @ksummers92

kennethjiang, 2020-07-29:
Thanks again folks for helping me figure this out. Releasing 1.3.3 that hopefully will fix this issue.