TheSpaghettiDetective/OctoPrint-Obico #67: sudo causes issue with Alpine Linux

fredrikbaberg, 2020-06-26:
I'm running OctoPrint in Alpine Linux (which itself may be a bit of a special case). When restarting OctoPrint with The Spaghetti Detective plugin enabled, I noticed it seems to rely on `sudo` which is not available in Alpine Linux by default.

Is the `sudo` part required for the plugin to operate? This seems to be the line causing issues for me:
https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/e686414d0ad3a4a6a42cb43a61684fbfd756546c/octoprint_thespaghettidetective/webcam_stream.py#L164

kennethjiang, 2020-06-27:
Are you running Alpine on a Pi? If so that'd explain it.

We have assumed that when OctoPrint runs on a Pi, the OS is Raspbian and hence `sudo` will be available.

fredrikbaberg, 2020-06-28:
In short: yes, it´s Alpine running on a Pi.
To be very specific I have OctoPrint running as a Home Assistant Addon on a Raspberry Pi 3B+.

I just had a closer look at the code I referenced and saw that the code path is specific for Pi. If the code is written assuming Raspbian I will probably encounter more issues than just `sudo`, one thing is that Alpine does not use the `service` command (part of `systemd`?).

Now that I know the assumption I hope to figure something out, I assume you are not ready to re-write the code to avoid both `sudo` and `service` just for this case. Right now I'm without a camera, hopefully I'll have a Pi camera by the end of next week so I can start testing myself.

kennethjiang, 2020-06-28:
I definitely can rewrite this part if we have a universal way that works for all flavors of Linux.

The fallback option will be to test if it's running on a Raspbian OS. And skip all these commands specific to Raspbian when the test returns false.

fredrikbaberg, 2020-07-06:
Sorry for the delay. Last week I received the camera so I could test it and right now it seems to work.

I don´t have a good suggestion how to rewrite it to be universal, a fallback option may be a simpler workaround.
For now I'll close this issue.

closed 2020-07-06 by fredrikbaberg