TheSpaghettiDetective/OctoPrint-Obico #68: Repeatedly being asked to install 1.26

timur-tabi, 2020-07-05:
I got a notification to install the 1.26 update, so I did.  The server rebooted, and I was once again asked to install 1.26.  I figured maybe the install didn't quite go through, so I said yes again.  The server rebooted, and I was asked a third time to install 1.26.  This time I clicked "ignore", but I'm quite certain that this process would have repeated indefinitely otherwise.

kennethjiang, 2020-07-05:
Thank you for reporting this problem to us. It was caused by the name change of our plugin.

There is no easy way to "fix" it. We just implemented a workaround to "force" OctoPrint to adopt the new name. When you see the prompt of upgrade to version 1.3.0, please do so, and the problem should be solved. Let me know if you still have the same problem after upgrading to 1.3.0.

RaymondHimle, 2020-07-16:
@timur-tabi Has this been resolved?

timur-tabi, 2020-07-16:
I guess so.  I've updated the plugin a few times now and have not had the problem.

On a side note, could you please rename the plugin back to its previous name?  "Access Anywhere - The Spaghetti Detective" is long and obnoxious.  I know why you did it, but I don't like it.  Nothing wrong with just calling it "Spaghetti Detective".

closed 2020-07-16 by RaymondHimle