TheSpaghettiDetective/OctoPrint-Obico #69: 404 in setup promt when behind proxy?

fredrikbaberg, 2020-07-07:
I'm running OctoPrint behind HAProxy. When going through the setup for TheSpaghettiDetective in OctoPrint some resources do not load.

Could it be the case that everything after the hostname (or IP:port) is dropped?

Example from console:
```
Failed to load resource: the server responded with a status of 404 (Not Found) http://<url:port>/plugin/thespaghettidetective/static/img/logo-square.png
```
Same with e.g. `/api/plugin/thespaghettidetective`.

The issue is that I access OctoPrint through `http://<url:port>/<generated id>`, but the `<generated id>` part seems to be dropped.

### Setup

The setup I have is a bit complicated, right now I don't have a simplified way to recreate it.
I'm running [Home Assistant](https://www.home-assistant.io/) on a Raspberry Pi 3B+. In Home Assistant I have an addon for OctoPrint, see https://community.home-assistant.io/t/repository-octoprint/22883.
Access to OctoPrint is through [Ingress](https://developers.home-assistant.io/docs/add-ons/presentation/#ingress). In practice this means that the URL used access OctoPrint is `http://IP:PORT/774437fd_octoprint/`, where `774437fd_octoprint` is an ID for the addon (example, different ID for each addon).

fredrikbaberg, 2020-07-07:
I have the possibility to expose OctoPrint directly if needed, so there is a workaround for me, but I think the behavior should be considered a bug.

kennethjiang, 2020-07-07:
Thank for reporting this issue. You are right that this bug is caused by the assuming that octoprint is running from the root of the domain, such as this: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/94f7b234fd136df3e3ba7a385d88115b16a05e6a/octoprint_thespaghettidetective/static/js/TheSpaghettiDetective.js#L10

Should we use relative path as opposed to absolute path that starts with "/"? Or is there an API in OctoPrint to re-construct a fully-qualified URL? Not sure if I have the time to do the research now so I'll really appreciate it if someone can research it and point us in the right direction, or better yet, submit a PR. :)

kennethjiang, 2020-07-09:
Closed with #70 . Thank you @fredrikbaberg

closed 2020-07-09 by kennethjiang