TheSpaghettiDetective/OctoPrint-Obico #73: IFTTT integration

apbarratt, 2020-08-19:
When the detective detects something fishy, or indeed feels the need to pause the print, I get an array of notifications.  Unfortunately, if these happen after 22:00, I don't see them as my phone has switched itself to Do Not Disturb mode.  I also sometimes don't look at my phone regularly enough to respond right away.

I would like to be able to use IFTTT to set up my own notification/response methods with other internet of things items.

Examples:
IF the TSD finds something fishy, THEN set Philips Hue bedside light orange at 1%.
IF the TSD pauses the print, THEN set Philips Hue bedside light to red at 50%.
IF the TSD detects a successful completed print, THEN turn down the heating in the print room and activate robotic vaccum cleaner.
IF the TSD detects something, THEN trigger a web request to my raspberryPi to make a servo motor turn and start communicating the detected event using semaphore flags so that passing ships are aware of the situation with the 3D printer.

kennethjiang, 2020-08-20:
Really good suggestion. And thank you so much for telling us a specific use case. We may not have resource to tackle this right now but I'm hoping someone will pitch in to make it happen soon. Otherwise we will need to wait until we wrap up the things we are currently working on (quite exciting!).

matthewschroer, 2021-01-30:
In a similar vein using MQTT would allow Home Assistant to set up similar automation. My use case would be I like to have "pause print" off when I'm awake but have it on when I'm asleep. So I could use Home Assistant to turn on and off this parameter at set times.

kennethjiang, 2021-01-31:
This is actually a good use case.

More and more users are telling us to formalize API/webhook integration. We will put that on our roadmap!