TheSpaghettiDetective/OctoPrint-Obico #74: Issue: Need to use Octolapse snapshots instead of standard Octoprint snapshots

pssnyder, 2020-08-30:
Problem Statement: I am using Octolapse to capture snaps of my print therefore I have positioned the camera in a way that the standard timelapse does not properly capture images of the full bed area. This results in TSD not seeing the print properly and it makes the monitoring and Focused Feedback somewhat inaccurate.

Question: How does TSD capture timelapse video? If I change the settings for Octolapse to output mpg video and still save it to the standard Octoprint location, will TSD pull the Octolapse videos instead?

Examples:
Standard Timelapse Capture
[image]

Octolapse Capture:
[image]

RaymondHimle, 2020-08-31:
It has to do with the machine learning model being based on images spread 10 seconds apart. TSD doesn't use any setting from the octoprint timelapse asides from camera URLs. As for using the octolapse snapshots there is no way for us to do so. You can add another camera that offers a better view and use that for TSD instead.

kennethjiang, 2020-09-01:
@pssnyder Assuming I didn't misunderstand your problem statement, I changed the title.

It is a good idea to give users the flexibility to set a location (or a neighborhood) for TSD to take snapshot, to maximize how much the model can be seen by the camera.

We can even further simplify the user experience by directly talking to octolapse if it's installed to take the snapshot at the same time as octolapse, assuming this is an optimal position to do so. However this extra dependency is quite hacky and brittle so I'm less convinced this is a good move.

In any case, this is not a small improvement and it may take a while before we get around to it.

pssnyder, 2020-09-04:
@RaymondHimle I do have a few IP based web cams that I could set up in a better position. How would I tell TSD which camera stream to use? Can I set a custom snapshot feed?

@kennethjiang Yea I can guess dependencies to integrate directly with Octolapse would become a nightmare. I do like the idea however of location based snapshots vs time based, which could really just be triggered by reading the GCode, which I assume TSD can gain access to. Then the only caveat would be figuring out a user friendly way for TSD users to provide ideal snapshot locations that actually coincide with locations that the print bed will be at some point, or at least get things in the "neighborhood" as you put it. Cool stuff. Looking forward to seeing if this gains any traction.

kennethjiang, 2020-09-05:
Glad that user experience seems to make sense to you. We are thinking about redoing the printer wizard. Will possibly incorporate this in the new wizard.

closed 2025-06-22 by pssnyder