TheSpaghettiDetective/OctoPrint-Obico #78: URL for TSD plugin api call is missing {printer_id} when proxied

kennethjiang, 2020-09-02:
Steps to reproduce:

1. Load proxied OctoPrint
2. TSD plugin api call fails in dev tools because it's URL is missing {printer_id} (screenshot).
[image]

3. Open TSD plugin settings in proxied Octoprint. In "advanced settings", it shows the account is not eligible for premium streaming, which is actually caused by the failed API call. (screenshot)
[image]

4. Open TSD plugin settings directly in Octoprint and go to the same place. The premium streaming is shown.

kennethjiang, 2020-09-14:
I thought it was fixed but just realized the issue is still there.

encetamasb, 2020-09-15:
@kennethjiang it works if tunnel url ends with /