TheSpaghettiDetective/OctoPrint-Obico #89: Unable to access advanced tab

jagr200, 2020-11-12:
Running TSD on Octoprint 1.4.2 on Pi 4b. Using my own server to watch the cameras. I am unable to access the advanced tab on TSD in this latest update. I used to be able to but I just installed another instance of Octoprint for another printer and I can not setup those settings in advanced.

kennethjiang, 2020-11-15:
Sorry somehow I missed your issue.

There is a known issue that advanced tab can't be show in wizard. But usually it shows in OctoPrint settings. So in your situation it doesn't even show in the settings?

kennethjiang, 2021-01-25:
1.5.0 rewrote the settings page. There is no more "advanced" tab".

closed 2021-01-25 by kennethjiang

gstevenson, 2021-01-25:
@kennethjiang  Where were the fields that were in the advanced tab moved to? I've just done a local install and now looking how to get to field to enter my IP

kennethjiang, 2021-01-26:
Sorry for the delay. It's moved here.

[image]