TheSpaghettiDetective/OctoPrint-Obico #90: Feature Request: Don't monitor preheat and cooldown periods

vrelk, 2020-11-17:
I have done a few prints today that have lasted a total of 30-60 seconds each (small washers for electrical isolation), but once you include the preheat and cool-down periods it was 14-15 minutes. TSD ended up watching for the entire time when the printer wasn't doing anything for most of it.

Can we get a way to stop watching during preheat and cool-down? There is no point in watching it, and it does use up detective hours for no reason.

kennethjiang, 2020-11-17:
Ideally OctoPrint provides a signal/event for the start of print head movement. I can't find it in its document but I know the vast majority of those cool OctoPrint functions are undocumented. Please comment to let us know if these is such signal if anyone is familiar with this part of the OctoPrint code.

vrelk, 2020-11-17:
So, I have been talking in Discord about this, which is where @kennethjiang suggested that I make this.

I also talked with SleepyBlueCat and it seems it might be part of the default G-code Cura uses for the printer I'm using at the moment... (The 5 minute delay that is most likely to speed up the cooling process)

Slicer: Cura
Printer: Monoprice Select Mini V2 (E3D)
Full g-code file it did it on: https://gist.github.com/vrelk/87a58c8794d7ca0ae5fc92603cf98263
Start & End G-Code: https://gist.github.com/vrelk/f667d499eee41c3d870976d376f1ca92
Examples: https://youtu.be/xQdbf2ZtvN0 https://youtu.be/TUcmTq23VqM

kennethjiang, 2020-11-19:
If I understood you correctly, you gave an example from a particular version of a particular slicer (Cura in this case).  However I believe this issue is about making this feature work for the majority of the slicers our users may use.