TheSpaghettiDetective/OctoPrint-Obico #91: Remove embedded Youtube video in setup tab

vordenken, 2020-12-04:
Since installing your plugin I get a lot of blocked domains from my adblocker because apparently the embedded youtube video in your plugins "setup" tab loads stuff every time I do anything in octoprint. I dont need to open the setup tab once but its still trying to load. These are the domains:

https://googleads.g.doubleclick.net/pagead/id
https://static.doubleclick.net/instream/ad_status.js
https://www.youtube-nocookie.com/generate_204?aSeGwg

I think this is a privacy issue and just totally unneccessary. Mabye just put a link to the video in the setup, no need for the video to be embedded.

Cheers,
VD

kennethjiang, 2020-12-06:
We will think about it.

gmergulhao, 2021-01-24:
I think this is also breaking other stuff. I cant setup the plugin since the css won't respond, hence I cannot change to the advanced tab nor does the finish button works

kennethjiang, 2021-01-25:
1.5.0 Removed the video link.

closed 2021-01-25 by kennethjiang