TheSpaghettiDetective/OctoPrint-Obico #101: TSD plugin attempts to access servers even when printer is turned off

timur-tabi, 2021-01-11:
It appears that the TSD plugin is access "servers" over the Internet, even when the the printer is inactive or even turned off.  I just received this error message:

> This window is to diagnose connection problems with The Spaghetti Detective server. It is not a diagnosis for your print failures.
> 
> The Spaghetti Detective plugin has run into issues. These issues may have prevented The Detective from watching your print effectively. Please check out our trouble-shooting page or reach out to us for help.
> 
> The plugin has failed to connect to the server 82 times (error rate 82%) since OctoPrint rebooted.
> 
>     The first error occurred at: Mon Jan 04 2021 06:38:36 GMT-0600 (Central Standard Time).
>     The most recent error occurred at: Mon Jan 11 2021 03:01:19 GMT-0600 (Central Standard Time).
> 
> Please check your OctoPrint's internet connection to make sure it has reliable connection to the internet.

Note the timestamp on the most recent error.  My router is programmed to reboot at 3am every night.  My printer has been turned off for the past couple days.  The only way TSD could notice that Internet was not available at 3:01AM is because it attempted to access the servers, even though no print is running!

The TSD plugin should notice that no print is active and should not make any attempt to access the Internet.

kennethjiang, 2021-01-11:
Hope this makes sense: https://discord.com/channels/614543405724205137/661327037360898063/798321998148927538

timur-tabi, 2021-01-13:
I understand why TSD might want to be available when the printer is not printing, but that's a feature I don't care about.  However, it makes no sense for TSD to do anything when the printer is not connected.  Then TSD can't do anything.

I would like an option to disable the features that required TSD to be communicating over the Internet when the printer is not connected and/or not printing.  I don't care about the "access anywhere" features of TSD.  I only care about the print monitoring features when it's actually printing.

kennethjiang, 2021-01-13:
"Connecting the printer" was also a commonly-asked feature before we made it available. If our plugin is not connected to the internet while the printer is not connected, by definition we can't make "Connecting the printer" feature work.

There are other commonly-requested features such as controlling the power supply to the printer. They are one step further beyond just "connected printer".

The outgoing (not incoming) connection from the plugin shouldn't have any security concerns if that's why you brought this up. If you are concerned about outgoing connection from your Pi, TSD cloud doesn't seem like a right fit for your needs to be honest. You can set up a TSD private server to achieve maximum security.

closed 2021-01-25 by raymondh2