TheSpaghettiDetective/OctoPrint-Obico #103: [BUG] Meta information about TSD is overlaying Octoprint UI

mhughes2k, 2021-01-25:
Following discussion on the Discord channel, I've found that the TSD meta information such as discord server are overlaid in the bottom left corner of the Octoprint UI:
[image]

My instance is running:
* Octoprint 1.5.3
* TSD 1.5.0
* UI customiser 0.1.2.5 (which when disabled causes this to *not* happen & disabling the "Improve Mobile / Responsive" option in UICustomiser had no effect)

This UI issue was present on my Octoprint instance on Brave Browser, Edge and Safari.

Exploring the outputted HTML it appears that 2 footer elements are being generated in the HTML:
[image]

ryryog25, 2021-01-25:
Same issue, HAR attatched
[octopi.local_Archive [21-01-25 13-04-21].zip](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/files/5869936/octopi.local_Archive.21-01-25.13-04-21.zip)

ryryog25, 2021-01-25:
Attached the entire page source here, in case anyone wanted to find the part shown in the screenshot by @mhughes2k 
[Octopi_HTML.txt](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/files/5869961/Octopi_HTML.txt)

mhughes2k, 2021-01-25:
Is it possible the <div> defined at 
https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/1500b65ef3d341e7fce1c6c9b86a17b2fefb83a3/octoprint_thespaghettidetective/templates/thespaghettidetective_settings.jinja2#L35

is getting confused with the .footer class that octoprint is using for the main navigation footer, as this code looks like its *meant* to be the footer *for the TSD settings* page.

mhughes2k, 2021-01-25:
If I change this from "footer" to "tsdfooter", issue appears to go away.

See https://github.com/mhughes2k/OctoPrint-TheSpaghettiDetective/pull/1/commits/a9ffad90c900c19f88756882d3c7f91b23c565ed

mhughes2k, 2021-01-25:
Looking into the UI Customiser code, I've found this section, which appears to be part that "upgrades" the footers to being fixed or not: https://github.com/LazeMSS/OctoPrint-UICustomizer/blob/ffc331700bbac02df997fb62b839263ba54f15e1/octoprint_uicustomizer/static/js/uicustomizer.js#L1671

It looks like it's doing a simple div.footer selection to determine what element is the *Octoprint* footer element. Unfortunately the TSD footer element *also* satisfies this CSS selector.

kennethjiang, 2021-01-26:
Fixed in https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/commit/43945c1d3593f20fd15e71d9761d86c3e64031e6

closed 2021-01-26 by kennethjiang