TheSpaghettiDetective/OctoPrint-Obico #104: [Feature] Option to set private server endpoint on the verification code page in setup wizard

kennethjiang, 2021-01-26:
**Is your feature request related to a problem? Please describe.**
The setup wizard, which initially launched, doesn't provide an option to set the endpoint to a private TSD server. And hence it can't be linked to a private server.

**Describe the solution you'd like**
Have an option to set the endpoint to a private TSD server in setup wizard.

**Describe alternatives you've considered**
The current workaround is to click "Finish" to dismiss the wizard. Set the endpoint in the settings, then "Re-run wizard" to finish the linking pro ess.

trowj, 2021-01-26:
Perhaps I'm misunderstanding the notes here, but I couldn't even set a custom URL by dismissing the wizard initially - the plugin section in settings didn't have anything except a logo. I had to connect my OctoPrint to a real TSD account/printer, then I could go into settings and change the URL and re-run the wizard.

kennethjiang, 2021-01-27:
@trowj This sounds like another bug. Are you able to reproduce the problem? If so, please provide the steps. Thank you so much!

kennethjiang, 2021-01-27:
Actually we found the bug and included the fix in 1.5.1 that we just released

kennethjiang, 2021-01-27:
Fixed in version 1.5.1

closed 2021-01-27 by kennethjiang