TheSpaghettiDetective/OctoPrint-Obico #111: [Feature] Add option to match the way that themify and other similar Octoprint plugins change the color scheme of Octoprint.

ryryog25, 2021-03-03:
One thing that has really been bugging me is that "themify" which I always run and makes Octoprint super cool looking (IMO) does not play nicely with TSD. 

For example, (and I have linked screenshots of all these below) the TSD plugin menu tabs just don't look good, they are white as that is the normal background of Octoprint, but with my setup (which is more of a gray background) it doesn't look anywhere near as sleek.

The Setup Wizard interestingly doesn't actually get switched from black to white text so it pops correctly (The plugin normally changes text so that it does in fact pop, and there are multiple different combinations of color schemes...) leaving the setup wizard basically impossible to read without straining your eyes.

After this the only thing that is truly broken in the Octoprint TSD plugin is the part where you enter in the 6-digit verification code and the background appears to be changed by the plugin, but the rest of the field, which I assume is not actually a part of the text field but rather a UI element, is white.

The white buttons are fine, I actually think they pop really well with the dark background, so I would almost say I don't want those removed, same for the white logos.

Everything else the plugin does a really good job of converting, so that would be it, unless I missed something.

It would be great if the TSD plugin could somehow adapt to these to make the UI flow as it does with vanilla Octoprint.

I don't really know how this would work, since I have a feeling those UI elements are just SVGs or something, so you would almost need to specifically create more SVGs that looked the same as themify's current settings and then be able to pull what it was set to.

SCREENSHOTS:
[image]
[image]
[image]

kennethjiang, 2021-03-10:
Thank you so much @ryryog25 for such a good bug report!

kennethjiang, 2021-03-21:
We fixed other issues except the tabs in the settings. We feel the default tabs don't really flow well with the rest of our settings page. So our designer added while border at the bottom to make it less strange in dark theme.

[image]

kennethjiang, 2021-03-21:
Fixed in 1.6.2

closed 2021-03-21 by kennethjiang

ryryog25, 2021-03-21:
This looks great! Having that extra bar does go a long way to make it look less strange with themify. I noticed that if I have the window smaller I get a UI that looks like this: 

[image]

I know this would affect the entire plugin UI and it is a rather selfish at this point, but just as an idea I wonder how it would look it this kind of style was kept all the time? I appreciate the work you have already done to fix this, just wanted to throw this in the air so at the very least it is out there, whether it is used or not.

Again, thank you for taking a look at this!

kennethjiang, 2021-03-24:
Bringing issues about product is never a selfish thing, as there are always other users who have the same problem.

Yes we made a decision decision to go simple when the screen is narrower, assuming most users use OctoPrint on the computer (or use Touch UI on the phone). Did you open this on an iPad?

ryryog25, 2021-03-26:
I use octoprint on my iPad a lot, but that particular screenshot was just a small browser window. I noticed that the TSD design did that when in a smaller form factor and was kind of wondering if there was a way to tell if themify was being used and use that design all the time? Idk, just a thought.