TheSpaghettiDetective/OctoPrint-Obico #118: libmmal.so is not available on arm64

dwery, 2021-04-01:
The plugin repeatedly tries to open libmmal.so, which is not available on arm64 and actually not needed if you are using an usb webcam.

kennethjiang, 2021-04-04:
sorry not sure why I missed this issue.

libmmal.so should be available on any good Raspbian system, including all OctoPi builds. It is an outstanding issue of the arm64 build that they are currently missing. That's why official OctoPi 0.18 doesn't ship the arm64 build.

Closing this as this is not a TSD plugin issue.

closed 2021-04-04 by kennethjiang

dwery, 2021-04-04:
@kennethjiang the plug-in can maybe stop retrying opening it. Even with an exponential back-off, it will never succeed and will just pollute the logs.

kennethjiang, 2021-04-04:
I'm not in favor of making a non-trivial change, which would increase the risk of bugs, because of 2 reasons:

- the scope of the impacted users are small: only the ones who have installed an arm64 build that is known to be effective (let me know if I'm wrong here).
- the user experience is not impacted. the only impact seems to be polluting the log files, which will be rotated out.

dwery, 2021-04-04:
The assumption here is that OctoPrint is only used on raspberries. Even if that were true, that library is useless unless you also use the integrated pi camera. Yes, logs are rotated, but having exceptions in them is not as clean as it could be.

kennethjiang, 2021-04-04:
Our code check if it's running on a Pi. It won't try to load Pi Camera module if it's not.

dwery, 2021-04-05:
This is what I'm seeing in the logs, on a Pi w/o that lib.
`octoprint.plugins.thespaghettidetective - ERROR - Backing off 68.966931 seconds: Janus quit! This should not happen. Exit code: 127`

is it related?