TheSpaghettiDetective/OctoPrint-Obico #128: ERROR - 'NoneType' object has no attribute 'get_metadata'

sylvaneau, 2021-07-23:
Hi,

I'm running last versions of TSD (both plugin and server) and the octoprint logs get flooded with this error message

`2021-07-22 17:20:18,167 - octoprint.plugins.thespaghettidetective - ERROR - 'NoneType' object has no attribute 'get_metadata'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_thespaghettidetective/print_event.py", line 76, in get_file_metadata
    file_metadata = plugin._file_manager._storage_managers.get(origin).get_metadata(path)
AttributeError: 'NoneType' object has no attribute 'get_metadata'`

Any idea of what's going on ?

Thx for your help

kennethjiang, 2021-07-25:
Sorry for the delay.

This error happens when the plugin can't retrieve the z-height about the gcode. I'm still trying to figure out when this will happen. Were you printing a gcode from an SD card when this happened to you?

In any case, the TSD server is fault-tolerant when z-height is not available. So it shouldn't impact any real function except you can't see the z-height progress of a print job.

sylvaneau, 2021-07-31:
Hi,

Yes I was printing from the SD card, I actually always print from the SD as i'm not really confident in relying on a third party app (octoprint in that case) to send g-code to the printer. Yes i know it's probably stupid ^^

kennethjiang, 2021-07-31:
Ok so this is a known issue in OctoPrint. Does it cause any problem in the app, except that the z-height is shown as '-'?

sylvaneau, 2021-08-01:
No it doesn't, I was investigating on another issue and I saw these errors in the logs, that's why I posted here.

Thx

Télécharger Outlook pour Android<https://aka.ms/AAb9ysg>

kennethjiang, 2021-08-01:
Ok I'll close this issue.

closed 2021-08-01 by kennethjiang