TheSpaghettiDetective/OctoPrint-Obico #132: [Feature] Use premium streaming on XYZ platform with hardware encoding

kunalgrover05, 2021-08-08:
**Is your feature request related to a problem? Please describe.**
TSD today only supports premium streaming using Raspberry pi hardware encoding. I am on a different machine which supports hardware encoding, I would like to use premium streaming. 

**Describe the solution you'd like**
Premium H264 streaming support for other platforms like: Intel / AMD on-board graphs, Nvidia graphs etc. 

**Describe alternatives you've considered**
Using 0.1 FPS streaming supported today. It is too inefficient for both detection and viewing. 

**Additional context**
NA

kunalgrover05, 2021-08-08:
I am proposing a docker image which would support premium streaming on any platform which has FFMPEG Hardware encoding support. It currently **breaks Octoprint** because of the different input expected by Octoprint but it should be possible to also add FFServer streaming to fix it back: http://trac.ffmpeg.org/wiki/ffserver

```
version: '2.4'
 
services:
  octoprint:
    image: octoprint/octoprint
    restart: unless-stopped
    ports:
      - 80:80
      - 8090:8080
    volumes:
     - octoprint:/octoprint
    devices:
     - /dev/ttyUSB0:/dev/ttyUSB0
    environment:
     - JANUS_SERVER=<Your Hostname>
  
  janus:
    restart: unless-stopped
    # Uses host networking because it uses a lot of ports for UDP streaming. 
    network_mode: host
    image: thespaghettidetective/janus:ubuntu
    command: sh -c "/opt/janus/bin/janus"
 
   ffmpeg:
      # Replace with your image with the appropriate HW acceleration build
      image: jrottenberg/ffmpeg:4.1-vaapi
      devices:
        # Your camera goes here
        - /dev/video2:/dev/video0
      # 8004 is the port in the Janus plugin config
      command: "-f v4l2 -framerate 25 -stats -i /dev/video0 -c:v libx264 -preset ultrafast -f rtp rtp://server:8004?pkt_size=1300"

volumes:
  octoprint:

```

B-0-B-B-Y, 2021-09-04:
I second this, would be a really nice feature to have implemented, I run a lot of printers off of a home server running multiple octoprint docker containers, and having premium camera streaming work on other platforms that aren't a RasPi would be great to see. @kunalgrover05 I'm curious to know, when you say currently what you propose "breaks Octoprint", can you describe what exactly breaks in Octoprint? Do you mean the entire octoprnt client doesn't work, just the camera feed from the "Control" tab in Octoprint doesn't work or etc? An alternative I have in mind is simply hosting an IP based camera feed, and just passing that IP address over to The Spaghetti Detective server (for when you have everything running over LAN this would make things very simple and nice to setup), and having the TSD server use the direct camera feed from there? That might be easier as there are plenty of pre-existing multi camera hosting server solutions, that can again be spun up as docker containers, which would make this a very nice, self contained little cluster with a lot of flexibility around how you run your cameras

kennethjiang, 2021-09-04:
Whatever reason this issue was missed from my radar. Sorry about that. Will get better at github hygiene.

The `docker-compose.yml` is for local dev environment only. It is ignored when a TSD end-user installs TSD plugin. So even if we merge your proposed change to this repo, it won't benefit our end users. But you can definitely clone this repo and make the change yourself.

Let me know if I misunderstood what you mean here.

kunalgrover05, 2021-09-06:
>  I'm curious to know, when you say currently what you propose "breaks Octoprint", can you describe what exactly breaks in Octoprint? Do you mean the entire octoprnt client doesn't work, just the camera feed from the "Control" tab in Octoprint doesn't work or etc? 

Yes. 
Octoprint follows a completely different model of using MJPG streaming only. In the ideal world, FFMPEG can easily support multiple outputs with FFServer outputting MJPG stream in parallel. However, FFServer(https://trac.ffmpeg.org/wiki/ffserver) is deprecated, so we need to figure out an alternative streaming mechanism for viewing in Octoprint. 

> It is ignored when a TSD end-user installs TSD plugin. So even if we merge your proposed change to this repo, it won't benefit our end users.

Yes, my goal is to only support this for Private server to begin with. We can probably extend it to users of the OctoPrint plugin customers in the future

kennethjiang, 2021-09-06:
> Yes, my goal is to only support this for Private server to begin with. We can probably extend it to users of the OctoPrint plugin customers in the future

Sounds good. Feel free to send a PR and I'll merge it.

B-0-B-B-Y, 2021-09-09:
> Yes.
> Octoprint follows a completely different model of using MJPG streaming only. In the ideal world, FFMPEG can easily support multiple outputs with FFServer outputting MJPG stream in parallel. [...]

You can supply an external IP address for the webcam stream in Octoprint, that's how I have my current cameras setup. I was suggesting we forego messing with Octoprint and how it handles webcam streaming for webcam devices passed into the Octoprint image, and instead modify The TSD plugin so that it can use that external IP camera address camera feed too. I'm not too sure what the fastest / easiest way to do that would be, but my initial thoughts are either creating a docker image that will act as the camera server + FFMPEG or creating a plugin for Octoprint that you can install directly inside of Octoprint that would take over the default camera mechanism, though I want this to be performant and Python is one of the last picks I would go for if performance is important, so I'm leaning more towards a custom docker image, maybe written in golang / rust.

What I am not sure about is how TSD Octoprint plugin handles / what it does with the camera feed from octoprint and how it sends the video information over to the TSD server, and then how does the TSD server feed that video information into the ML container.

I am not sure this makes complete sense, so want to give an example run through of what I am suggesting, let me know if you think this is going to be harder / more time consuming than what you are proposing.

USB Camera -> Connected to a Server (Let's call this server X) that streams video from said USB camera over the LAN. e.g. as long as you're connected to the same LAN network as server X, and server X is actively streaming, if you navigate to server X's IP, for the sake of this let's say it's 192.168.1.10:[whatever port you want to assign, can default to just using 80]

All server X is doing is a raw video output, much like the video output you get from the default Octoprint /webcam/?action=stream

In Octoprint, you change your webcam url to server X's IP and port, so in this example we change it to:
http://192.168.1.10

At this point, if we press Test cam in Octoprint, we should get the nice camera video feed. Then, a modified version of the TSD Octoprint plugin can use that same camera feed and send it over to the TSD server. All processing in the ML container and the web container regarding video remains the same, except now we get more than 1 frame every 10 seconds, aka an actual video feed again on the TSD server side.

Does that sound doable or is it too far out? Like I said I have no idea how the TSD Octoprint plugin + TSD Server communicate between each other when it comes to the video feeds, so this could be an impossible idea.

kennethjiang, 2021-09-10:
@B-0-B-B-Y Thank you for the detailed proposal. This is actually already doable with compatibility mode set to "always". But there is some downside with compatibility mode, primarily around the high cpu and low resolution and/or framerate resulted from low encoding efficiency. For more info: https://www.thespaghettidetective.com/docs/streaming-compatibility-mode/

Dids, 2022-03-01:
The core issue still seems to be that there's currently no way to offload the TSD video encoding process, at least not without moving OctoPrint itself off of a Pi etc.

If the encoder could run elsewhere, this would allow for much more flexibility, but also improved performance, as it wouldn't necessarily have to be constrained by the Pi anymore, let alone a Pi that's also controlling a 3D printer.

encetamasb, 2022-03-02:
> The core issue still seems to be that there's currently no way to offload the TSD video encoding process, at least not without moving OctoPrint itself off of a Pi etc.
>
> If the encoder could run [...]

How you might be able to do it:

(0) In theory you could use an external janus server (on non rpi-32bit platforms) by setting JANUS_SERVER environment variable for octoprint process, but that does not work well with tsd cloud. So to make work (1-3) bellow, you still need an rpi with 32bit os to host octoprint.

(1) Disable video streaming in tsd plugin settings, so the builtin video pipeline does not start.

(2) Open port 8004 from the outside on the rpi and comment out this line https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.plugin.streaming.jcfg.template#L144 so it can listen on all interfaces

(3) Send h264 encoded video stream as udp to port 8004; for an usb camera, you can see ffmpeg params here https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/webcam_stream.py#L196 
or you can use gstreamer pipleine, something like here https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/gst/run_gst.sh#L11 You cannot use these samples as is, you will need to build your solution yourself.