TheSpaghettiDetective/OctoPrint-Obico #139: [BUG] TSD doesn't respect camera_raspi_options in /boot/octopi.txt, other weird port 8080 issues

Gelob, 2021-10-14:
**Note**

**Describe the bug**
I'm running octopi 0.18 on a Raspberry Pi 3 Model A Plus Rev 1.0 using a Raspberry Pi NoIR Camera V2. By default octoprint sets the white balance to auto which results in a redish image on a NoIR camera.
[image]
 In order to fix this I have to use a patched version of mjpg_streamer (https://github.com/guysoft/OctoPi/issues/666) and then set the following settings in /boot/octopi.txt.
```
camera="raspi"
camera_raspi_options="-awb greyworld"
```

[image]
This works fine when TheSpaghettiDetectivePlugin is disabled, and my image is no longer red because the greyworld white balance option is working. If I enable TheSpaghettiDetective plugin, this stops working and my image goes back to being red. Also I notice that with TSD plugin disabled mjpg_streamer is listening on 127.0.0.1:8080 only, however when TSD plugin is enabled octoprint is now listening on port *.:8080 instead of just localhost.

```
octoprint 4059      pi   36u  IPv4  60212      0t0  TCP *:8080 (LISTEN)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Set options in /boot/octopi.txt with TSD disabled, restart octopi, and see color be correct.
2. Enable TSD plugin an restart octopi, see wrong color again

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] MacOS 11.6
 - Browser [e.g. chrome, safari] Chrome 
 - Version [e.g. 22] 94.0.4606.71 (Official Build) (x86_64)

kennethjiang, 2021-10-15:
This is a "known issue". Please check out this help doc: https://www.thespaghettidetective.com/docs/streaming-compatibility-mode/#when-should-i-always-stream-in-the-compatibility-mode

Let me know things will start to work for your once the plugin is configured to "always stream in compatibility mode".

Gelob, 2021-10-15:
Thanks, I missed that and it did fix it What about port 8080 being exposed on all interfaces instead of just loopback, is that intentional?