TheSpaghettiDetective/OctoPrint-Obico #153: [BUG] Streaming issue that's possibly caused by MTU issue

kennethjiang, 2022-01-23:
A user reported choppy premium streaming when the device is on a 5G cellular network with 130 Mbps down and 20 Mbps up.

The Pi is connected via ethernet with wifi disabled.  Currently getting ~850 Mbps down and 35 Mbps up from the pi.

The premium streaming works well on the local network.

In his log file, there are many lines of:

```
"2022-01-23 12:26:39,100 - octoprint.plugins.thespaghettidetective - DEBUG - JANUS: [ERR] [ice.c:janus_ice_outgoing_traffic_handle:4054] [5130086522468657] ... only sent -1 bytes? (was 1310)"
```

That error message is definitely related in some way.  it starts going crazy and generating that same message a couple hundred times a second only when I try to view the stream from my phone over 5G, but not anywhere else.

NegussugeN, 2022-01-24:
Just some more random pieces to the puzzle: I very hastily pieced together a PP2P OpenVPN setup with the server hosted on an old Macbook I had lying around that resides on the same network as the Octopi.  After installing and connecting to the VPN on my Pixel 6 Pro over 5G, the quality of the stream was immediately improved.  I didn't bother setting up split tunneling when configuring the VPN, so even though I was only getting 20 Mbps down over the VPN connection, the choppiness of the stream was 100 times better than with the VPN disconnected and getting about 120 Mbps down over 5G.

kennethjiang, 2022-01-24:
> Just some more random pieces to the puzzle: I very hastily pieced together a PP2P OpenVPN setup with the server hosted on an old Macbook I had lying around that resides on the same network as the Octopi. After installing and connecting to the VPN on my Pixel 6 Pro over 5G, the quality of the stream was immediately improved. I didn't bother setting up split tunneling when configuring the VPN, so even though I was only getting 20 Mbps down over the VPN connection, the choppiness of the stream was 100 times better than with the VPN disconnected and getting about 120 Mbps down over 5G.
> 
> Update: I've been using a Pi 4b for all of this. Just to rule out some kind of hardware issue, I also swapped my Octopi SD card from the 4b to a 3b+ but the results were the same.

Thanks for the info. This can be another evidence that low-level networking issues such as the MTU is at play here.

NegussugeN, 2022-01-24:
Here's something that's interesting: If the room is very dimly lit, then the stream performs fairly decent on High (1640x922) settings over 5G.  I believe that makes sense because, as I understand it, when it is dark less data is being transmitted.  

I tried (somewhat blindly) adjusting the payload size up and down found in the janus.py file, but that didn't really have any effect, so I reverted it back to the default of 1500.  I also tried playing with the bitrate found in the webcam_stream.py, but that didn't make much of a difference either.

If I drop the settings to Medium (960x540) the stream is reasonably smooth with only a few hiccups.  While I suppose I could just stick with Medium, I'd also like to use the camera to record decent quality timelapses, so that won't really work.  There's definitely something else at play here, as all of the devices in the equation have more than enough bandwidth/speed to be able to stream at 1640x922 at 25fps.

encetamasb, 2022-01-24:
@NegussugeN Hi! I assume it's not an issue with your OS or code, as there is a network configuration when it works properly (over VPN, right?).

(1) Anyway, if you want to play with the MTU, you need to change it here (assuming advanced streaming):

https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/gst/run_gst.sh#L11

`... ! rtrtph264pay mtu=1200 ! ...`

MAX_PAYLOAD_SIZE in janus.py is related to the datachannel.

(2) Another possible change in the gst pipeline could be to lower the bitrate or how `omxh264enc` is configured. See another thread here https://groups.google.com/g/meetecho-janus/c/PBv1jlfdqmk .

(3) You can try this too, but I don't believe its going to help. https://gitlab.freedesktop.org/gstreamer/gst-plugins-good/-/issues/549 There are other report on the broad internet, where changing these limits helped.

Me I think this is a carrier issue.

encetamasb, 2022-01-24:
Ah, If you use the raspicam, then gstreamer is not in use. Do you have an usb camera?

NegussugeN, 2022-01-24:
> @NegussugeN Hi! I assume it's not an issue with your OS or code, as there is a network configuration when it works properly (over VPN, right?).

> Me I think this is a carrier issue.

When trying to troubleshoot, everything was working fairly smoothly over 5G when I established a VPN connection to another device on the same network as the Pi.

As far as a carrier issue, I sent my shared stream link over to @kennethjiang on discord the other day and he confirmed that it was choppy on his end as well.

NegussugeN, 2022-01-24:
> Ah, If you use the raspicam, then gstreamer is not in use. Do you have an usb camera?
> 
> _MTU_ in case of raspicam can be altered here (I assume)
> 
> https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/webcam_stream.py#L181

Thanks for the info.  I'll play with this number and see if anything changes.

encetamasb, 2022-01-24:
> Thanks for the info. I'll play with this number and see if anything changes.

And just for the record,  you can change nat config of janus here: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.jcfg.template#L180

NegussugeN, 2022-01-24:
This seems to be related to the error I'm experiencing, but it doesn't really seem to lead anywhere: https://groups.google.com/g/meetecho-janus/c/PBv1jlfdqmk

I tried playing with the MTU for raspicam, both raising and lowering it, but the error messages in the log remain the same (spamming that same message in the original post 100s of times a second) and the choppiness is the same.

I tried playing with the nat settings in the janus.jcfg.template file.  I really have no idea what each of these settings do, so I just went through them one by one after a reboot.  With full_trick=true, I didn't really see any difference.  With ice_lite and ice_tdcp set to true, the stream stopped working unless I was on the same network.  Any suggestions on things I can change or edit within the config file?

It seems odd that I would be the only one experiencing this issue, as I didn't do any kind of custom installation or compiling of my own version.  I just downloaded it from the Octoprint plugin manager.

NegussugeN, 2022-01-24:
Another thing to note, as I mentioned earlier, I've also tried this sd card in a Pi 3b+, to no avail.  Additionally, I reformatted the SD card and did a fresh install of OctoPi and TSD before reaching out on Discord.

NegussugeN, 2022-01-28:
I'm still trying to find something that would make my scenario unique. I know it's a stretch but is it possible that placing this Arducam low distortion sensor/lens is somehow not fully supported with TSD?  Because it's a bigger lens, is it somehow pulling in more data than what is allowable/expected by TSD? Here is what I'm using currently on an official pi v2 camera: https://www.arducam.com/product/arducam-imx219-low-distortion-m12-mount-camera-module-drop-in-replacement-for-raspberry-pi-v2-camera/

encetamasb, 2022-01-28:
I don't think that it's the camera. This must be a network related problem, what might be not a bug on your side but .. . My guesses:

(1) Router configuration vs our webrtc pipeline 
(2) Carrier vs our webrtc pipeline 
(3) Something with your local network (interference...? if dimming lights fixes it)

I used to have issues with my celluar network, but for many months it has been working pretty well for me.

Please find me on discord (same username), I would be happy to check your stream myself @NegussugeN

encetamasb, 2022-01-28:
> I'm still trying to find something that would make my scenario unique. I know it's a stretch but is it possible that placing this Arducam low distortion sensor/lens is somehow not fully supported [...]

Could you try with an usb camera? Streaming from raspicam uses a different video pipeline, it's possible that some difference creates an issue at driver level.

NegussugeN, 2022-01-28:
Well now... This is interesting.  I have IPv6 disabled on my network at the router level.  @encetamasb came to the conclusion that might be the problem right at the same time.  Sure enough, I changed the following line to false to disable IPv6 and the stream is _substancially_ better than it was before.  There's still a decent amount of dropped frames, but definitely manageable.  I know next to nothing about the way Janus works, so I really have no idea why not utilizing IPv6 would cause this.

kennethjiang, 2022-01-29:
@NegussugeN Super interesting! I initially thought this was caused by an MTU issue. But wondering how ipv6 will play a role here. Routing? Or still things related to packet limit? @encetamasb Any ideas?

encetamasb, 2022-01-29:
@NegussugeN can I ask you to test it with ipv6 enabled in the router? 

I guess it's a bug in Janus or coTURN.

NegussugeN, 2022-01-30:
I just went through and enabled IPv6 on my network/router (which wasn't super fun, as my pihole doesn't like to play nice with IPv6).  I rebooted everything, make sure that the pi running Octoprint was assigned an IPv6 address, switched the IPv6 flag back to true within the janus.jcfg.template file, and the problem is the same as before.  Very choppy stream.  I then went to just set it back the way it was and disable IPv6 at the router and change the janus.jcfg.template flag back to false.  After a reboot, same behavior.  I ended up editing my sysctl.conf to completely disable IPv6 and after a reboot, everything is working great again.  I'm really not sure what the issue is, or if it is even on my end, but it appears that any time IPv6 is remotely in the picture in my given setup, the stream goes back to being laggy.

encetamasb, 2022-01-30:
@NegussugeN Thank you for testing it. In my case all network providers (cable, celluar) provide an ipv6 address, and my devices use it automatically without any problems. Anyway, I guess we need to make ipv6 support optional in the plugin, and maybe set it off by default. We will decide soon.

Another user reported slowness during stream initialization, 15s+. Switching off ipv6
solved the issue for him too.

kennethjiang, 2022-01-30:
@NegussugeN Can you help us do some more testing? Please uninstall TSD plugin (keep data) and install a testing version using this URL: `https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/archive/refs/heads/noipv6.zip`

Then do 2 tests with it:

1. when ipv6 is disabled in your router.
2. when ipv6 is enabled in your router.

Let us know how the tests go.

kennethjiang, 2022-01-30:
@NegussugeN We have decided to set ipv6 to "False" in janus.jconfig.template, as janus' ipv6 implementation seems to be buggy. Once the next plugin version is available, you can update it.

closed 2022-01-30 by kennethjiang