TheSpaghettiDetective/OctoPrint-Obico #164: [BUG] TSD fails to control webcam service on Arch ALARM

LMBernardo, 2022-04-12:
## Steps to reproduce the behavior:
1. Install TSD plugin on an Octoprint instance running on Arch ALARM
2. Restart Octoprint server
3. Watch logs

#### OS: Arch Linux ALARM - 5.15.32-3-rpi-ARCH

## The Problem
The TSD plugin attempts to stop/start the webcam service using `sudo service webcamd stop/start`, which is not a command that exists on Arch Linux. This is handled by the following lines of code:
https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/379fd59f017e2efcf491e52ac54c5c797e2337f8/octoprint_thespaghettidetective/webcam_stream.py#L141
https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/379fd59f017e2efcf491e52ac54c5c797e2337f8/octoprint_thespaghettidetective/webcam_stream.py#L188
https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/379fd59f017e2efcf491e52ac54c5c797e2337f8/octoprint_thespaghettidetective/webcam_stream.py#L308

Everything seems to work just fine when running the webcam stream separately, so this is not a critical issue. It does mean that TSD has no control over starting / stopping the webcam stream though.
It looks like `uname()` is already pulled in [`get_tags()`](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/379fd59f017e2efcf491e52ac54c5c797e2337f8/octoprint_thespaghettidetective/utils.py#L144), which could be used to check for Arch Linux:
```
>>> import platform
>>> platform.uname()
uname_result(system='Linux', node='****', release='5.15.32-3-rpi-ARCH', version='#1 SMP Fri Apr 1 13:05:11 UTC 2022', machine='armv7l')
```

I believe '-ARCH' in the version is unique to Arch Linux. When I have time to set up a fork of TSD I may attempt a PR for this.
In the meantime...

## The Workaround
The main point of filing this issue is to share a workaround for any other ALARM users out there. The following works for me:

### Step 1
Create `/etc/systemd/system/webcamd.service`, or rename existing webcam service to "webcamd.service"

E.g.,
`~$ cat /etc/systemd/system/webcamd.service`
```
[Unit]
Description=Run webcam stream
After=network.target
After=network-online.target

[Service]
ExecStart=mjpg_streamer -i "input_uvc.so" -o "output_http.so"
User=root
Group=root
Type=exec
Restart=on-failure
RestartSec=30

[Install]
WantedBy=multi-user.target
```

### Step 2
Create `/usr/sbin/service`
`~$ sudo touch /usr/sbin/service`
`~$ sudo chmod +x /usr/sbin/service`

This is just a shell script that maps from `service ACTION UNIT` to `systemctl UNIT ACTION`, e.g.,
```
#!/bin/YOURSHELLHERE
systemctl "$2" "$1"
```

### Step 3
TSD appears to install a sudoers config file at `/etc/sudoers.d/octoprint-service`, with the following content:
`~$ cat /etc/sudoers.d/octoprint-service`
```
pi ALL=NOPASSWD: /usr/sbin/service
```

So if your Octoprint instance runs under the user 'pi' you can skip this step.

If not, you'll need to add these commands to your `sudoers` file. For example, if Octoprint runs under user / group `octoprint`:
```
octoprint ALL=NOPASSWD: /usr/sbin/service webcamd stop,/usr/sbin/service webcamd start
```

### Step 4
Restart your Pi and make sure the errors are gone.

kennethjiang, 2022-04-15:
Thank you so much for this detailed bug report.

Did you receive [the premium streaming](https://www.thespaghettidetective.com/docs/streaming-compatibility-mode/) after the workaround was implemented? The premium streaming is very specific to Raspbian. So if the workaround won't make the premium streaming work on Arch Linux, the right fix is to make sure we don't try to start the premium streaming unless we are sure the OS is Raspbian.

LMBernardo, 2022-04-17:
> Thank you so much for this detailed bug report.
>
> Did you receive [the premium streaming](https://www.thespaghettidetective.com/docs/streaming-compatibility-mode/) after the workaround was [...]

I don't think the premium streaming is working, unfortunately, however this may have more to do with my use of Python 3.10 than anything else.

The plugin status page reports "Webcam streaming: Premium (compatibility)," however I can't see the webcam when not printing and I am getting the basic 1 frame every ten seconds or so.

I'm not sure if this is related to Arch ALARM itself or just to my particular configuration, but (I believe) the relevant log lines are these:
```
[15431]: 2022-04-16 14:55:44,965 - octoprint.plugins.thespaghettidetective - INFO - Starting webcam streamer
[15431]: Exception in thread Thread-68 (start):
[15431]: Traceback (most recent call last):
[15431]:   File "/usr/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
[15431]:     self.run()
[15431]:   File "/usr/lib/python3.10/threading.py", line 946, in run
[15555]: pam_systemd_home(sudo:account): systemd-homed is not available: Unit dbus-org.freedesktop.home1.service not found.
[15555]: octoprint : PWD=/ ; USER=root ; COMMAND=/bin/service webcamd stop
[15555]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=973)
[15431]:     self._target(*self._args, **self._kwargs)
[15431]:   File "/usr/lib/octoprint/lib/python3.10/site-packages/octoprint_thespaghettidetective/janus.py", line 85, in start
[15431]:     ensure_janus_config()
[15431]:   File "/usr/lib/octoprint/lib/python3.10/site-packages/octoprint_thespaghettidetective/janus.py", line 53, in ensure_janus_config
[15431]:     with open(janus_conf_path, "wt") as fout:
[15431]: PermissionError: [Errno 13] Permission denied: '/usr/lib/octoprint/lib/python3.10/site-packages/octoprint_thespaghettidetective/bin/janus/etc/janus/janus.jcfg'
```

Additionally, I get a whole bunch of the following errors while printing. This is almost certainly a Python 3.10 issue. Just adding it here for context:
```
[1304]: AttributeError: 'NoneType' object has no attribute 'get_metadata'
1304]: 2022-04-08 01:26:20,080 - octoprint.plugins.thespaghettidetective - ERROR - 'NoneType' object has no attribute 'get_metadata'
[1304]: Traceback (most recent call last):
[1304]:   File "/usr/lib/octoprint/lib/python3.10/site-packages/octoprint_thespaghettidetective/print_event.py", line 80, in get_file_metadata
[1304]:     file_metadata = plugin._file_manager._storage_managers.get(origin).get_metadata(path)
```

It would probably be more helpful if someone else with an ALARM setup, but who is running an older version of Python, could check and see if they are able to use Premium streaming with this workaround.

kennethjiang, 2022-04-18:
So I believe the right fix is change the code to test if TSD is running in a Raspbian and if not, disable the premium streaming (janus) altogether.

@encetamasb Is this is something you can do?

The `get_metadata` is more of an OctoPrint issue. I have struggled to understand why somethings OctoPrint returns None for its store manager. But it shouldn't matter as the only impact this error has is you won't see the z-height in the app.