TheSpaghettiDetective/OctoPrint-Obico #167: [Feature] Manually Archive Printer -- Or -- Un-archive causes archiving of oldest active printer

tyeth, 2022-05-29:
Hi, I'd love to be able to use the printers more like site-licenses back in the day, i.e. I pay for two printers, but I have 3 (4 if you include a decomissioned mendel). Most of the time the main beast is lasering so no need, but it has the larger build surface so semi-regularly gets swapped back to being active. I'd like to be able to manually archive printers and then unarchive if my license count allows.

Currently: 
  To use specific printers with License_Count < Printer_Count you must delete printers to all me to Un-Archive / swap a printer. Then to swap back I must re-add and setup the deleted printer, which will auto-archive the swapped one.

Desired:
  I can Archive any active printer and it is added to the Archived Printers list and stops consuming a license. I can then un-archive as normal.

tyeth, 2022-05-29:
I think the system automatically archiving the printer which was least recently used would work well for print farms assuming you have those kind of customers. Standard error message if they are all actively printing. I can imagine manually working out which printers to archive would be a pain with more than ten.

tyeth, 2022-05-29:
Also just realised this is in the wrong forum 🤦 
@kennethjiang sorry

kennethjiang, 2022-05-30:
Implemented in https://github.com/TheSpaghettiDetective/obico-server/commit/87ba39579e1e4e010b80f2f953bbc8f88c3161b1 . Should be available in the next server release, probably in a few days.

closed 2022-05-30 by kennethjiang

tyeth, 2022-05-31:
Oh my goodness, thank you so much!